kantlivelong/OctoPrint-PSUControl #1: Error to start

PhenomenCan, 2017-01-26:
Hi, I test it OctoPrint 1.3.0 and 1.3.1 and same error to start plugin. I read the logs and see this error:

File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_psucontrol/__init__.py", line 86, in initialize GPIO.setup(self.senseGPIOPin, GPIO.IN) RuntimeError: No access to /dev/mem. Try running as root!

Plugin not activate

kantlivelong, 2017-01-26:
Are you running on a Raspberry Pi?
If so, are you using OctoPi or did you install OctoPrint manally?

kantlivelong, 2017-01-26:
After looking a bit further. Could you also run the following bash command while OctoPrint is running and paste the results.

`groups $(ps aux | grep -i 'octoprint' | grep -v 'grep' | awk ' { print $1 }') && ls -la /dev/gpiomem`

KriLL3, 2017-01-28:
Same issue here, was excited for this plugin since I've been trying to do similar functionality to this badly in g-code, I installed octoprint quite some time ago using the ISO from the site and that command gives me this:

http://i.imgur.com/X4iGtFx.png

kantlivelong, 2017-01-28:
See https://raspberrypi.stackexchange.com/questions/40105/access-gpio-pins-without-root-no-access-to-dev-mem-try-running-as-root

I'll look into adding some detection for this case.

kantlivelong, 2017-02-02:
I've confirmed that the details link above resolve the issue. Upcoming release will log the errors instead of forcing the plugin to unload.

closed 2017-02-02 by kantlivelong