kantlivelong/OctoPrint-PSUControl #2: Provide example on how to configure navbar button

tfohlmeister, 2017-01-27:
Great plugin! Could you please provide a quick example on how to install that nice power / status button on the top nav as shown on your picture?
https://github.com/kantlivelong/OctoPrint-PSUControl/blob/master/psucontrol_navbar_settings.png

Thanks!

tfohlmeister, 2017-01-27:
I have set up the plugin as you described and it seems to run fine (at least there are messages in my log). *However,* the green status button with the bolt doesn't show up in my top nav bar. Do I need to setup this button separately?

kantlivelong, 2017-01-27:
You shouldn't need to do anything for it to appear provided everything is working properly.

- OctoPrint Version?
- Are you using OctoPi?
- When your power supply is on are you seeing 3.3v after your divider?
- Could you provide DEBUG logs following these steps:
- - Enable DEBUG loggin ( http://docs.octoprint.org/en/master/configuration/logging_yaml.html ) [Use octoprint.plugins.psucontrol for the component name]
- - Stop OctoPrint
- - Clear OctoPrint Logs
- - Start OctoPrint
- - Wait a minute or so.
- - Switch the PSU on/off (Wait 30 seconds or so in between. Physically switch if needed)
- - Attach logs

Thanks

tfohlmeister, 2017-01-27:
OctoPrint version 1.3.1, yes I am using OctoPi on a Raspberry B+.
I have a 5V connector on my control mainboard which I connected to **pin 8** on my Raspberry. I use a relay board to drive my power supply which is active low (see #3). This might be an issue, because switching on/off is reversed.
DEBUG logging file:
[octoprint.log.txt](https://github.com/kantlivelong/OctoPrint-PSUControl/files/735957/octoprint.log.txt)

kantlivelong, 2017-01-27:
The log indicates that it is indeed reading pin 8. (PS: I hope you used a voltage divider on that 5V. You might kill the Pi otherwise)

If you are not seeing the lightning bolt at all I am wondering if it is a browser issue or OctoPrint 1.3.1 (I'm still on 1.3.0)

If you are using Chrome would you be able to open the developer console (CTRL+Shift+J), select the console tab and enter the following and let me know at what point the indicator appears (If it does):
`$("#powercontrol_psu_indicator").css('color', 'red');`
`$("#powercontrol_psu_indicator").attr('class', 'fa fa-wrench');`
`$("#powercontrol_psu_indicator").css('visibility', 'visible');`

kantlivelong, 2017-01-27:
I Think I've fixed the indicator issue along with issue <https://github.com/kantlivelong/OctoPrint-PSUControl/issues/3>. Would you be willing to install the devel version to test? Make sure that you uninstall the existing plugin first.

URL: 

> https://github.com/kantlivelong/OctoPrint-PSUControl/archive/devel.zip

Paste the URL into plugin manager(Example screenshot below):
[image]

tfohlmeister, 2017-01-28:
First off: Thanks for your excellent support and your hint regarding the 3.3V input limit. I was dead sure the Raspi could handle 5V and had to look it up. I didn't fry my Pi but you were right about 5V being out of range. Anyway, I found a 3.3V pin on my mainboard so this shouldn't be an issue anymore.

Secondly: I installed your dev version as instructed and it works **perfectly** now. The button shows up on top and the inverse functions works as intended! Thank you so much! As far as I can tell, this version is good to be released.

closed 2017-01-28 by tfohlmeister