kantlivelong/OctoPrint-PSUControl #4: Error with auto on/off

copystring, 2017-01-30:
2017-01-30 18:46:59,381 - octoprint.plugins.psucontrol - INFO - Auto-On - Turning PSU On (Triggered by M190)
2017-01-30 18:46:59,383 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2017-01-30 18:46:59,484 - octoprint.util.comm - ERROR - Error while processing hook psucontrol for phase queuing and command M190 S85:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.1-py2.7.egg/octoprint/util/comm.py", line 2113, in _process_command_phase
    hook_result = hook(self, phase, command, command_type, gcode)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_psucontrol/__init__.py", line 208, in hook_gcode_queuing
    self.turn_psu_on()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_psucontrol/__init__.py", line 229, in turn_psu_on
    self.check_psu_state()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_psucontrol/__init__.py", line 117, in check_psu_state
    r = GPIO.input(self.senseGPIOPin)
ValueError: The channel sent is invalid on a Raspberry Pi

Waiting for it to turn off doesn't work either. I haven't logged that though. I guess that's the same issue.

kantlivelong, 2017-01-30:
Hello,

* What model Pi are you using?
* What pin # is being used for sense?

copystring, 2017-01-30:
Hi, I'm using a Pi 3 B.
Pin is on 0. I'm using the command option to switch. Do I need to define a pin for that?

kantlivelong, 2017-01-30:
Yes. See https://github.com/kantlivelong/OctoPrint-PSUControl/blob/master/README.md

copystring, 2017-01-30:
Hm, OK. I might misunderstand how this plugin works. This doesn't work with GCode alone?
My PSU is connected to pin PS_ON on my ramps 1.4 and is switchable with M80 and M81.

kantlivelong, 2017-01-30:
It currently requires a sense pin to confirm if the PSU is on/off instead of assuming based on last command sent. I may make the sense pin optional in the future.

If you are using an ATX supply you can either connect a 3.3v wire with a resistor to a pin of choice or use 5v/12v with a voltage divider.

copystring, 2017-01-30:
Oh, I see. Having that optional would be better in my opinion. :)
I'm a bit of a noob when it comes to resistors even though I have a hole bunch of them. Just never used them.
What kind of resistor would I need on the 3.3v wire?

kantlivelong, 2017-01-30:
Anything between 1-10k would be fine.

copystring, 2017-01-30:
Thanks. Used a 10k resistor. It turns on fine now. It doesn't seem to turn off though. I've set the timeout for testing to 1 minute.

kantlivelong, 2017-01-30:
Were any hotends on? It will wait for hotends to be <= 50C(default) before powering off. Also any command sent that is not in the ignore list will abort the power off action

copystring, 2017-01-30:
No, nothing was on. Everything is ~20°C. Apart from M105 nothing was sent after I've manually sent G28 (which turned the PSU on just fine).

kantlivelong, 2017-01-30:
Can you send fresh DEBUG logs?

copystring, 2017-01-30:
Sure.

`2017-01-30 20:59:19,930 - octoprint.plugins.psucontrol - INFO - Auto-On - Turning PSU On (Triggered by G28)
2017-01-30 20:59:19,931 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2017-01-30 20:59:19,932 - octoprint.events - DEBUG - Firing event: PowerOn (Payload: None)
2017-01-30 20:59:19,934 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x6aa52cb0>>
2017-01-30 20:59:20,037 - octoprint.events - DEBUG - Firing event: Home (Payload: None)
2017-01-30 20:59:20,045 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x6aa52cb0>>
2017-01-30 20:59:38,117 - octoprint.events - DEBUG - Firing event: PositionUpdate (Payload: {'reason': None, 'e': 0.0, 't': 0, 'f': None, 'y': 0.0, 'x': 0.0, 'z': 11.8})
2017-01-30 20:59:38,118 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x6aa52cb0>>`

kantlivelong, 2017-01-30:
Could you please attach the entire log from startup?

copystring, 2017-01-30:
Sorry. Here it is: http://pastebin.com/BmJdb5xd

kantlivelong, 2017-01-30:
It doesnt look like DEBUG logging is on for psucontrol.

- Could you provide DEBUG logs by following these steps:
- - Enable DEBUG logging ( http://docs.octoprint.org/en/master/configuration/logging_yaml.html ) [Use **octoprint.plugins.psucontrol** for the component name]
- - Stop OctoPrint
- - Clear OctoPrint Logs
- - Start OctoPrint
- - Wait a minute or so.
- - Switch the PSU on
- - Wait 1-2 minutes past the expected idle timeout.
- - Attach logs

copystring, 2017-01-30:
Alright. Done all that. Here's that log: http://pastebin.com/seLQpuX8

kantlivelong, 2017-01-30:
It doesn't look like it is seeing a high value on the sense pin(21) when turning the PSU on. The lightning bolt icon in the navbar should also turn green when on. 

First thing to check is to ensure you are specifying the right pin number. This plugin uses physical pin instead of BCM pin numbering. For the Pi 3 see [this diagram](https://s-media-cache-ak0.pinimg.com/originals/84/46/ec/8446eca5728ebbfa85882e8e16af8507.png). By specifying pin 21 you will be using GPIO09(SPI_MISO).

If that is correct then you would want to verify that you see 3.3V after the resistor when the PSU is switched on and 0v when off.

copystring, 2017-01-30:
Man. I'm so sorry :D
I've never used the GPIO pins of the Pi before. Didn't realize that GPIO 21 is not pin 21. I've set it to 40 and it works. Sorry for the inconvenience. Works perfectly now.

kantlivelong, 2017-01-30:
No problem. Glad its working :D

closed 2017-01-30 by kantlivelong