kantlivelong/OctoPrint-PSUControl #5: GPIO conflict

MoonshineSG, 2017-02-08:
Due to the presence of `GPIO.cleanup()` in the initialise sequence causes other plugins that use GPIO and have already been initialised to fail. Basically clearing any other GPIO those need to use. 

And there's also a conflict with GPIO.BOARD vs GPIO.BCM on other plugin. Try/except would be needed?

kantlivelong, 2017-02-08:
Will take a look at improving cleanup.

Might be slightly tricky to fix the BOARD/BCM issue but I'll review the RPi.GPIO source to see if there's any possible way to detect.

Which plugin is this conflicting with?

MoonshineSG, 2017-02-08:
One of my own. Switch.
No hurry on my side. I decided not to use your plugin, but instead to improve on mine. :) 
But maybe there are other GPIO plugins that might cause problems.

kantlivelong, 2017-02-13:
I'm wondering if it would make sense to have a global option to specify BOARD vs BCM for GPIO. Though I'm not sure as to what the best route for implementing it would be. 

The simplest idea i can think of is a "Raspberry Pi GPIO" plugin that simply sets BOARD/BCM and have other plugins check the value. Don't think it's possible to specify another plugin as a dependency though.

Kunsi, 2017-02-13:
`GPIO.cleanup()` should never be used, unless you're absolutely sure you're the only one dealing with GPIO pins. Just set up the pins you need.

This also interferes with my plugin (to be submitted) to display print information on a connected LCD display.

Regarding BOARD vs. BCM: Most GPIO tutorials on the internet i found use board numberings, because they're much easier to simply count, than having to get some sheet to map board numbers to BCM numbers.

kantlivelong, 2017-02-13:
@Kunsi 

I can see why using `cleanup()` with no arguments could be bad but using `cleanup(pin)` should be okay.

See I find BOARD to be easier as I don't need to look at anything but the header and know what pin is used. That's why I think some global option should be used so that the user can decide.

Kunsi, 2017-02-13:
Am 13.02.17 um 22:52 schrieb Shawn Bruce:
> I can see why using |cleanup()| with no arguments could be bad but using
> |cleanup(pin)| should be okay.

Yes, that should be perfectly fine, because it will not mess up other
scripts.

> See I find BOARD to be easier as I don't need to look at anything but
> the header and know what pin is used. That's why I think some global
> option should be used so that the user can decide.

Yes, I also prefer BOARD numberings.

MoonshineSG, 2017-02-13:
No inter plugin dependency on OctoPrint. 
A global  setup and set either bcm or board and put this in the plugin documentation. Maybe a setting for the user to choose board or bcm whicever they prefer. 
I'll ask the developer if she's ok to do this in OctoPrint.

kantlivelong, 2017-02-22:
I've pushed fixes for cleanup() and setmode() to devel.

The plugin will internally keep track of configured pins and call cleanup() for each pin individually.

For setmode() it will configure for BOARD if no mode is set. Otherwise it will use an internal lookup table to convert BOARD to BCM.

Feedback appreciated.

kantlivelong, 2017-03-15:
Confirmed working in devel.

closed 2017-03-15 by kantlivelong