kantlivelong/OctoPrint-PSUControl #7: PSU Control is not ACL aware

synman, 2017-02-09:
I discovered by happen chance today that PSU Control is not protected by any authentication / authorization scheme provided by OctoPrint.

This means that if your server is accessible to the outside world and you have PSU Control installed that an anonymous user can power on/off your printer.  This presents a significant challenge for internet connected instances of OctoPrint that rely on authentication to gain administrative access to OctoPrint and its plugins.

[image]

kantlivelong, 2017-02-10:
Looks like it's more of a UI issue as the server does in fact reject the request to turn on/off if not authenticated.

Fixed in https://github.com/kantlivelong/OctoPrint-PSUControl/tree/issue-7. Will be merged to master when ready.

synman, 2017-02-10:
nice... thank you for the follow up and aesthetic fix.  it caught my eye last night while I was in the middle of print.

kantlivelong, 2017-02-13:
Fixed in 9c10364e6ae0b0b3e63a543331fe11176db257e9

closed 2017-02-13 by kantlivelong

synman, 2017-02-14:
thanks for the quick fix