kantlivelong/OctoPrint-PSUControl #8: PSU Control does not play nicely with TouchUI

synman, 2017-02-10:
I'll likely open this as an issue in the TouchUI repo as well but there is currently a problem between TouchUI and PSU Control.  When on a low density touchscreen (3.5 LCD attached to the Pi) with PSU Control and TouchUI active when you try to open the System/Settings/Logon spanner (see attached) the entire context menu is blanked out.

You can also simulate this fairly easily by just shrinking your browser's width (see attached).

[image]

synman, 2017-02-10:
And tying back to TouchUI issue:

https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/97

synman, 2017-02-10:
here you can see what is going on:

[image]

kantlivelong, 2017-02-11:
Fixed in TouchUI - https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/97

closed 2017-02-11 by kantlivelong