kantlivelong/OctoPrint-PSUControl #9: 5v instead of 3.3v?

efrainregalado, 2017-02-11:
Not quite an issue but more of a feature request:

I have my PSU connected to a relay box where the gpio signal for on is 5v instead of 3.3v, 0v is off.
Is there a way to configure/alter the code to send this voltage instead?

I currently use this command to turn it on from bash (ran from the config.yaml):
   gpio export 17 out; gpio -g write 17 0
And using wiringpi on python: 
    wiringpi.pinMode(17, 1)
    wiringpi.digitalWrite(17, 0)

Thanks

kantlivelong, 2017-02-11:
The GPIO on a Raspberry Pi is only 3.3v. If it works from command line it should work with the plugin.

efrainregalado, 2017-02-11:
Wow, that was a quick response.
Thanks for the heads up, i'm still new to the whole thing. After a bit of fiddling i got it running - i was using the BCM pin instead of the physical one. I also needed to check the invert box.
The one thing that doesn't seem to be happy is the sensing bit, it shows it inverted. But without sensing seems to work great.

I've been using commands and delays in the config.yaml to turn it off after a print and save energy. This is much better, thanks for sharing it!

kantlivelong, 2017-02-11:
Ideally the sensing pin should be connected to one of the output voltages on your power supply that the plugin is switching on/off. In my case I connect the sensing pin to one of the 12v pins from my ATX supply using a voltage divider. The sensing pin should receive 3.3v when on and 0v when off.

efrainregalado, 2017-02-11:
Got it, that makes sense. I was using the same pin that sends the signal to the relay box.

closed 2017-02-11 by efrainregalado