kantlivelong/OctoPrint-PSUControl #10: [Request] Add ability to dis-/reconnect when switching PSU

Kunsi, 2017-02-13:
My PSU does also supply power to the printers mainboard. That means octoprint will display 'connection timeout' when PSU switches off, and i have to manually reconnect when switching PSU on.

It would be nice if your plugin would handle that as well, including an (ideally configurable) delay between 'PSU On' and 'reconnect to printer' to allow firmware to boot up.

kantlivelong, 2017-02-13:
Have you considered trying [PortLister](http://plugins.octoprint.org/plugins/portlister/)?

Kunsi, 2017-02-13:
That indeed would solve the issue when powering on, yet still would result in a connection timeout when PSU switches off.

kantlivelong, 2017-02-13:
I'll see if it's something I can add in the future.

If you are using an ATX supply I would recommend connecting the printer PCB & Pi to your +5VSB if possible.

Kunsi, 2017-02-13:
I indeed use an ATX supply. However the Microcontroller on the printers PCB is powered by an voltage regulator, which is powered by the 12V rail of the printer - so no way to fix that without changing electronics.

Kunsi, 2017-02-13:
disconnecting should be possible with calling `self._printer.disconnect()` before actually switching off in https://github.com/kantlivelong/OctoPrint-PSUControl/blob/master/octoprint_psucontrol/__init__.py#L267

kantlivelong, 2017-04-09:
Available in 0.1.4

closed 2017-04-09 by kantlivelong