kantlivelong/OctoPrint-PSUControl #11: [Request] Add checkbox for internal pullup/pulldown

Kunsi, 2017-02-13:
It should be possible to switch on the internal pullup/pulldown-resistors, which would allow for direct wiring of ATX PSUs.

kantlivelong, 2017-02-13:
Only pins configured as an input support pull up/down. What you are looking for is **open drain**. which the Raspberry Pi does not support.

To switch on an ATX supply via GPIO use a standard NPN transistor.

See [Using the Raspberry Pi to control an ATX power supply](http://www.instructables.com/id/Using-the-Raspberry-Pi-to-control-an-ATX-power-sup/)

closed 2017-02-22 by kantlivelong

Kunsi, 2017-02-22:
Yeah, also figured ATX PSUs do have an internal pullup to 3V3, so it should be possible to connect the PSU directly to GPIO.