kantlivelong/OctoPrint-PSUControl #13: Doesn't seem to work, GPIO problem

Naesstrom, 2017-02-21:
I installed this plugin today but nothing seems to happen with my relay, checking the logs I get this at the boot:
`2017-02-21 22:25:47,217 - octoprint.plugins.psucontrol - INFO - Running RPi.GPIO version 0.6.3
2017-02-21 22:25:47,218 - octoprint.server - ERROR - Error while trying to migrate settings for plugin psucontrol, ignoring it
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.1-py2.7.egg/octoprint/server/__init__.py", line 293, in run
    settings_plugin_config_migration_and_cleanup(implementation._identifier, implementation)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.1-py2.7.egg/octoprint/server/__init__.py", line 284, in settings_plugin_config_migration_and_cleanup
    implementation.on_settings_initialized()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_psucontrol/__init__.py", line 95, in on_settings_initialized
    self._configure_gpio()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_psucontrol/__init__.py", line 109, in _configure_gpio
    GPIO.setmode(GPIO.BOARD)
ValueError: A different mode has already been set!`

And when I try to start it with G0 and later using the icon in the top _(would it be possible to have that up there from boot?)_ I get this in the log:
`2017-02-21 22:27:05,286 - octoprint.plugins.psucontrol - INFO - Auto-On - Turning PSU On (Triggered by G0)
2017-02-21 22:27:05,287 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2017-02-21 22:27:05,288 - octoprint.plugins.psucontrol - ERROR - The GPIO channel has not been set up as an OUTPUT
2017-02-21 22:27:11,309 - octoprint.plugins.psucontrol - INFO - Switching PSU Off
2017-02-21 22:27:11,310 - octoprint.plugins.psucontrol - ERROR - The GPIO channel has not been set up as an OUTPUT
2017-02-21 22:27:13,044 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2017-02-21 22:27:13,045 - octoprint.plugins.psucontrol - ERROR - The GPIO channel has not been set up as an OUTPUT`

Running Version 1.3.1 (master branch) on a Rpi3, relay connected like the image attached, the on_led is dimly lit all the time, if I connect the Signal wire to the 3.3V pin the relay turns on and the on_led shines brighter so the relay is working.
[image]

kantlivelong, 2017-02-21:
Thanks for the detailed info.

Are you using any other plugin or application that uses GPIO?

Naesstrom, 2017-02-22:
Hi, only filament sensor (https://github.com/MoonshineSG/OctoPrint-Filament) but not connected since I'm waiting for the optical sensor. I can uninstall that and try again

Naesstrom, 2017-02-22:
Ok, uninstalled the other plugin and now I get this in the log at boot:
`2017-02-22 15:45:00,794 - octoprint.plugins.psucontrol - INFO - Running RPi.GPIO version 0.6.3
2017-02-22 15:45:00,795 - octoprint.plugins.psucontrol - INFO - Using GPIO for On/Off
2017-02-22 15:45:00,795 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 4
2017-02-22 15:45:00,796 - octoprint.plugins.psucontrol - ERROR - The channel sent is invalid on a Raspberry Pi`

Followed by this later when I use the on/off button
`2017-02-22 15:46:29,752 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2017-02-22 15:46:29,753 - octoprint.plugins.psucontrol - ERROR - The channel sent is invalid on a Raspberry Pi
2017-02-22 15:46:46,129 - octoprint.plugins.psucontrol - INFO - Switching PSU Off
2017-02-22 15:46:46,130 - octoprint.plugins.psucontrol - ERROR - The channel sent is invalid on a Raspberry Pi
2017-02-22 15:46:49,599 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2017-02-22 15:46:49,600 - octoprint.plugins.psucontrol - ERROR - The channel sent is invalid on a Raspberry Pi`

kantlivelong, 2017-02-22:
Pin 4 is 5v. Make sure you are using BOARD pin numbering.

[image]

Naesstrom, 2017-02-22:
yeah I'm using GPIO04, should I enter 7 at the "On/Off GPIO Pin" box? I thought that since it said GPIO in the label I should use the GPIO number not the physical number?

kantlivelong, 2017-02-22:
Correct. GPIO04 is board pin 7.

The GPIO library used offers the option of using BCM(GPIO#) or BOARD(Pin#) for specifying pin number and it does indeed cause some confusion/issues.

Kunsi, 2017-02-22:
The original error should be fixed by #5

kantlivelong, 2017-02-22:
It may be fixed depending on the order that the plugins(In this case psucontrol and filamentsensor) are loaded.

The pin number still needed to be correct though.

@Naesstrom you are welcome to try the devel branch once you confirm that it is working with pin 7.

Naesstrom, 2017-02-22:
Yup, now it works just fine! Thanks for the clarification regarding the correct number! What's the upside of the dev version?

kantlivelong, 2017-02-22:
devel branch currently has better cleanup(unconfigure) functionality for GPIO pins and will also internally work with BCM pin numbering if another plugin has already set the mode to BCM opposed to BOARD.

See #5 

I've done minimal testing so far but it appears to work. Feedback would be helpful though.

Naesstrom, 2017-02-22:
Testing it out is the least I can do for the great support you are giving, what's the easiest way to install the dev version on my Pi?

kantlivelong, 2017-02-22:
Uninstall the plugin then install via URL.

URL: https://github.com/kantlivelong/OctoPrint-PSUControl/archive/devel.zip

Paste the URL into plugin manager(Example screenshot below):
[image]

kantlivelong, 2017-02-24:
Going to close as the original issue is resolved.

If you have any issues with the devel branch please comment in #5.

closed 2017-02-24 by kantlivelong