kantlivelong/OctoPrint-PSUControl #19: gpio setup

mcmasterp, 2017-03-24:
When I run this with the enclosure plugin and a few others I get this error and non functioning behavior from psu control. I set the enclosure plugin option to use BOARD pins as that seemed to match your usage and would eliminate a pin conflict I thought. 

Here's a log file:
`2017-03-24 01:06:20,832 - octoprint.server - INFO - Starting OctoPrint 1.3.2 (master branch)
2017-03-24 01:06:20,833 - octoprint.server - INFO - ******************************************************************************
2017-03-24 01:06:21,049 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2017-03-24 01:06:25,430 - octoprint.plugin.core - INFO - Found 19 plugin(s) providing 15 mixin implementations, 14 hook handlers
2017-03-24 01:06:25,525 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2017-03-24 01:06:26,007 - octoprint.plugin.core - INFO - Found 19 plugin(s) providing 15 mixin implementations, 14 hook handlers
2017-03-24 01:06:26,017 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2017-03-24 01:06:26,081 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2017-03-24 01:06:26,520 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2017-03-24 01:06:40,173 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python -m pip" as command to invoke pip
2017-03-24 01:06:44,743 - octoprint.util.pip - INFO - Version of pip is 9.0.1
2017-03-24 01:06:44,745 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages, --user flag needed => no, virtual env => yes
2017-03-24 01:06:44,750 - octoprint.plugin.core - INFO - Initialized 15 plugin implementation(s)
2017-03-24 01:06:44,777 - octoprint.plugins.psucontrol - INFO - Running RPi.GPIO version 0.6.3
2017-03-24 01:06:44,778 - octoprint.plugins.psucontrol - INFO - Using sensing to determine PSU on/off state.
2017-03-24 01:06:44,778 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 19
2017-03-24 01:06:44,779 - octoprint.plugins.psucontrol - INFO - Using GPIO for On/Off
2017-03-24 01:06:44,780 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 3
2017-03-24 01:06:44,794 - octoprint.plugin.core - INFO - 19 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/plugins/announcements
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/plugins/corewizard
|  Cost Plugin (0.1.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_cost
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/plugins/cura
|  Detailed Progress Plugin (0.1.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_detailedprogress
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/plugins/discovery
|  EEPROM Marlin Editor Plugin (0.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_eeprom_marlin
|  Enclosure Plugin (2.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_enclosure
|  FileManager (0.1.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filemanager
|  GCODE System Commands (0.1.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_gcodesystemcommands
|  Gcodebar Plugin (0.1.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_GCodeBar
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/plugins/pluginmanager
|  PSU Control (0.1.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_psucontrol
| !Slic3r (1.0.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_slic3r
|  Slicer (1.0.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_slicer
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/plugins/softwareupdate
| !STL Viewer (0.3.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stlviewer
| !TouchUI (0.3.6) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_touchui
| !Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/plugins/virtual_printer
2017-03-24 01:06:44,797 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2017-03-24 01:06:44,810 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2017-03-24 01:06:44,815 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2017-03-24 01:06:44,846 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2017-03-24 01:06:45,388 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance "i2 Samuel XL" for _http._tcp
2017-03-24 01:06:45,409 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance "i2 Samuel XL" for _octoprint._tcp
2017-03-24 01:06:45,413 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance "i2 Samuel XL" for SSDP
2017-03-24 01:06:45,497 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2017-03-24 01:06:45,540 - octoprint.server - INFO - Listening on http://127.0.0.1:5000
2017-03-24 01:06:46,026 - octoprint.server - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://192.168.1.207/'}
2017-03-24 01:06:46,071 - octoprint.plugins.enclosure - INFO - Failed to read Temperature
2017-03-24 01:06:46,149 - octoprint.plugins.enclosure - INFO - Failed to read Humidity
2017-03-24 01:06:49,797 - octoprint.plugins.psucontrol - ERROR - You must setup() the GPIO channel first
2017-03-24 01:06:52,917 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.17
2017-03-24 01:06:54,826 - octoprint.plugins.psucontrol - ERROR - You must setup() the GPIO channel first
2017-03-24 01:06:55,769 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin 1.1.0-RC8 (Github)"
2017-03-24 01:06:56,237 - octoprint.plugins.enclosure - INFO - Failed to read Temperature
2017-03-24 01:06:56,298 - octoprint.plugins.enclosure - INFO - Failed to read Humidity
2017-03-24 01:06:58,676 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin 1.1.0-RC8 (Github)"
2017-03-24 01:07:01,223 - octoprint.plugins.psucontrol - ERROR - You must setup() the GPIO channel first
2017-03-24 01:07:06,305 - octoprint.plugins.psucontrol - ERROR - You must setup() the GPIO channel first
2017-03-24 01:07:06,413 - octoprint.plugins.enclosure - INFO - Failed to read Temperature
2017-03-24 01:07:06,470 - octoprint.plugins.enclosure - INFO - Failed to read Humidity
2017-03-24 01:07:11,312 - octoprint.plugins.psucontrol - ERROR - You must setup() the GPIO channel first
2017-03-24 01:07:16,314 - octoprint.plugins.psucontrol - ERROR - You must setup() the GPIO channel first
2017-03-24 01:07:16,530 - octoprint.plugins.enclosure - INFO - Failed to read Temperature
2017-03-24 01:07:17,636 - octoprint.plugins.enclosure - INFO - Failed to read Humidity
2017-03-24 01:07:21,317 - octoprint.plugins.psucontrol - ERROR - You must setup() the GPIO channel first
2017-03-24 01:07:26,319 - octoprint.plugins.psucontrol - ERROR - You must setup() the GPIO channel first
[... 102 lines cut ...]
2017-03-24 01:11:01,394 - octoprint.plugins.psucontrol - ERROR - The GPIO channel has not been set up as an OUTPUT
2017-03-24 01:11:01,495 - octoprint.plugins.psucontrol - ERROR - You must setup() the GPIO channel first
2017-03-24 01:11:02,701 - octoprint.plugins.psucontrol - ERROR - You must setup() the GPIO channel first
2017-03-24 01:11:07,704 - octoprint.plugins.psucontrol - ERROR - You must setup() the GPIO channel first
2017-03-24 01:11:10,293 - octoprint.plugins.enclosure - INFO - Failed to read Temperature
2017-03-24 01:11:10,348 - octoprint.plugins.enclosure - INFO - Failed to read Humidity
[... cut ...]
2017-03-24 01:11:40,634 - octoprint.plugins.enclosure - INFO - Failed to read Temperature
2017-03-24 01:11:40,688 - octoprint.plugins.enclosure - INFO - Failed to read Humidity
2017-03-24 01:11:42,742 - octoprint.plugins.psucontrol - ERROR - You must setup() the GPIO channel first
2017-03-24 01:11:47,747 - octoprint.plugins.psucontrol - ERROR - You must setup() the GPIO channel first
2017-03-24 01:11:50,746 - octoprint.plugins.enclosure - INFO - Failed to read Temperature
2017-03-24 01:11:50,801 - octoprint.plugins.enclosure - INFO - Failed to read Humidity
`

kantlivelong, 2017-03-24:
Would you mind trying the devel branch? I've added functionality to use BCM if another plugin has already setup GPIO with BCM. Unfortunetly though it won't help if psucontrol loads before another plugin that uses GPIO and uses BCM.

Adding option to select BOARD vs BCM soon.

mcmasterp, 2017-03-24:
sure what url to install dev?

kantlivelong, 2017-03-24:
https://github.com/kantlivelong/OctoPrint-PSUControl/archive/devel.zip

Just make sure that you uninstall the plugin before installing via URL.

mcmasterp, 2017-03-24:
ok I set it up. I reverted enclousre plugin option to BCM as well and it is working correctly (temp sensing aside) but psu control still no go.

logentries of interest:
`2017-03-24 01:42:09,486 - octoprint.plugins.psucontrol - INFO - Running RPi.GPIO version 0.6.3
2017-03-24 01:42:09,487 - octoprint.plugins.psucontrol - INFO - Using sensing to determine PSU on/off state.
2017-03-24 01:42:09,488 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 10
2017-03-24 01:42:09,489 - octoprint.plugins.psucontrol - INFO - Using GPIO for On/Off
2017-03-24 01:42:09,490 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 2
2017-03-24 01:42:09,490 - octoprint.plugins.psucontrol - ERROR - The channel sent is invalid on a Raspberry Pi
2017-03-24 01:42:09,500 - octoprint.plugin.core - INFO - 19 plugin(s) registered with the system:`

I'll test enclosure plugin in board mode now. brb!

kantlivelong, 2017-03-24:
Invalid channel would mean that an invalid pin was specified. Make sure that you specify BOARD pins.

mcmasterp, 2017-03-24:
well I tried all combos of both psu control setup and enclousre plugin setup to be sure and nothing works. the one way I can get it to work is after a boot where both plugins are set to use BOARD pins, pins set correctly, at first it psu control is not working but enclosure plugin is. If I now go into psu control settings, uncheck "Get PSU on/off state via GPIO", save settings, go back into settings and check to activate sensing, then save again. at this point the lightening bolt lights up and it begins correctly sensing but I see no signal on the on off pin i set

kantlivelong, 2017-03-24:
It looks like the enclosure plugin uses GPIO.cleanup() on start and on settings save. This essentially will wipe out the mode and all configured GPIO pins for any OctoPrint plugin.

Does psucontrol work without the enclosure plugin?

closed 2017-04-06 by kantlivelong