kantlivelong/OctoPrint-PSUControl #21: Triggering plugin from shell

Erutan409, 2017-03-29:
Is there a way to do this?  Reason I ask is because I use Printoid (Android app to connect to OctoPrint) and I'd like to be able to use this plugin with it, by using the app's shell function capabilities.

kantlivelong, 2017-03-29:
Not very familiar with Printoid. Does it just execute system commands on the OctoPrint system or do "shell functions" have some other meaning?

Erutan409, 2017-03-29:
Executing commands from the shell.  So, basically via SSH.  I don't plan on implementing the sensing GPIO.  So, I'd want to coordinate on/off via your plugin and not worry about a state not being noticed, using the printer interchangeably from the web GUI to my Android app.

Erutan409, 2017-03-29:
I could write a script that I could run against python from said app, but again - I don't want to lose track of its state based on the aforementioned.

kantlivelong, 2017-03-29:
You can create a shell script that calls the API. 

Example with curl:
`curl -s -H "Content-Type: application/json" -H "X-Api-Key: YOUR_API_KEY" -X POST -d '{ "command":"turnPSUOn" }' http://YOUR_OCTOPRINT_SERVER/api/plugin/psucontrol`

Available API commands:
turnPSUOn
turnPSUOff

Erutan409, 2017-03-29:
Awesome, thank you!

kantlivelong, 2017-03-29:
No prob.

closed 2017-03-29 by kantlivelong