kantlivelong/OctoPrint-PSUControl #22: Execute additional commands

azalanono, 2017-04-03:
Hello.
I notice that Marlin RC8 has a massive problem on my printer. The first printjob works fine, but all succeeding one have so much extrusion, that I must immediately abort, otherwise the nozzle srucks in 1mm thick filament...
I found out that disconnecting and re-connecting resets the Arduino. After that, the next printjob works again.

It would be great if OctoPrint-PSUControl could enforce a disconnect when turning off the ATX powersupply. Octoprint will then automatically re-connect, reset hereby the Arduino, and everything is fine.

kantlivelong, 2017-04-03:
The disconnect option is in the pipeline.  Not sure what the title has to do with the request though.

kantlivelong, 2017-04-03:
Duplicate of #10

closed 2017-04-03 by kantlivelong