kantlivelong/OctoPrint-PSUControl #23: PSU not switched on by PSUControl

Indystef, 2017-04-03:
Hi,

first of all: I think this is a great plugin, exactly what I needed. However - I cannot get it to work with GPIO. Hardware wise, I followed the transistor method described in [http://www.instructables.com/id/Using-the-Raspberry-Pi-to-control-an-ATX-power-sup/](url), using GPIO 29 for pulling the PSU_ON line low. This works fine from the command line, and also through a menu option I added to OctoPrint's system menu. However, the PSU is not turned on by PSUControl. Furthermore, if I add GPIO 28 as feedback, all prints start to be choppy, with filament blurbs in smaller features (looks like the communication with the controller can't keep up with the print).
The logs below are for an attempt of a print. PSUControl claims it turned the PSU on, but it did not. I then used the menu option to turn the PSU on (at 18:22:29,390). After the print I tried to turn the PSU off/on/off (using the green flash) - the logs don't show an error, but the PSU stayed on all the time.
Also (maybe important) - unless I manually set the mode of GPIO to out (through ssh), nothing works at all.

Any help is welcome.

`2017-04-03 18:21:33,246 - octoprint.plugins.psucontrol - INFO - Auto-On - Turning PSU On (Triggered by M140)`
`2017-04-03 18:21:33,247 - octoprint.plugins.psucontrol - INFO - Switching PSU On`
`2017-04-03 18:22:29,390 - octoprint.server.api.system - INFO - Performing command for custom:Printer on: gpio write 29 1`
`2017-04-03 18:28:53,828 - octoprint.util.comm - INFO - Finished in 378.068 s.`
`2017-04-03 18:34:12,918 - octoprint.plugins.psucontrol - INFO - Switching PSU Off`
`2017-04-03 18:34:18,382 - octoprint.plugins.psucontrol - INFO - Switching PSU On`
`2017-04-03 18:34:22,900 - octoprint.plugins.psucontrol - INFO - Switching PSU Off`

kantlivelong, 2017-04-03:
It looks you are specifying pin 28/29 using BCM numbering when you should be specifying as BOARD(physical).

Indystef, 2017-04-03:
Ah … works now, using pins 40 and 38. Btw. the BCM numbers would be 21 and 20 for GPIO 29 and 28 respectively.

Thanks,

Stefan

kantlivelong, 2017-04-03:
Too many pin numbering options. :open_mouth:

closed 2017-04-03 by kantlivelong