kantlivelong/OctoPrint-PSUControl #25: on/off switch not working after update

SaverBE, 2017-04-16:
After the latest update, the plugin is not working anymore for me.
I have set it up with GPIO pin 16 to switch my atx power supply, but after update i can't click the power button (lightning symbol) on octoprint.
Before the update the button would work (become green), but wouldn't switch power supply, i had to manually set a different number of pin, save, then go back in settings and set correct pin again, then save, then it would work.
But now i can't even click the button anymore, just stays greyed out.
Pls advice.

kantlivelong, 2017-04-16:
- Hardware?
- OctoPrint version?
- Could you provide fresh logs?

SaverBE, 2017-04-17:
1. raspberry pi 3 b, anet a8
2. octoprint 1.3.2
3. weird, i can't see any logs...

SaverBE, 2017-04-17:
Sorry, you can remove this thread.
Found more odd things checking logs and plugins, nothing was available, started googling on that, found an error within octoprint.
I logged out and back in, everything is fine now.

closed 2017-04-17 by kantlivelong