kantlivelong/OctoPrint-PSUControl #27: PSU won´t turn off when Idle, stuck in loop?

LundSoftwares, 2017-04-17:
Got an issue where my PSU won´t turn off when the Idle time is reached.
It keeps stuck in some loop waiting for heaters but everything is turned off and cooled below the set temp. This line keeps repeating in the standard Log: "Waiting for heaters(tool0) before shutting off PSU..." 

And in the Debug log it looks like this over and over:
"DEBUG - Heater tool0 = 187.3C
INFO - Waiting for heaters(tool0) before shutting off PSU..."

So, it seems to register the last known temp from the extruder, but it never changes and it´s stuck in the loop.

Suggestions?

Printer: Anet A8
Octoprint version: 1.3.2
PSUcontrol version: 0.1.4
Complete Debug Log:
[octoprint-log.txt](https://github.com/kantlivelong/OctoPrint-PSUControl/files/926719/octoprint-log.txt)

kantlivelong, 2017-04-17:
It looks like everything is working appropriately assuming tool0 is your hotend. As you can see in the log the temp is 187.3C and the plugin will wait until it is below 40C (idleTimeoutWaitTemp).

Did the UI report that your hotend was 187.3C at that time?

LundSoftwares, 2017-04-17:
Yes I only got one hotend so it´s tool0. 
Problem seems to be that when the Print job is done, Octoprint doesn´t keep polling the printer for temp´s. 
Looking at the printer display, it goes down = hotend and bed is turned off, but in octoprint the temps keeps their values they had seconds after the job is done. 

The PrintScreen is taken now, 30min after the job is done, Octoprint keeps reporting 187,3 degrees.
[image]

Is there a setting in Octoprint somewhere that I´we missed that keeps the temp polling alive?

kantlivelong, 2017-04-17:
If you send M105 in the terminal what gets reported back?

LundSoftwares, 2017-04-17:
Send: M105
Recv: ok 2
Recv: T:59.95 /0 B:24.44 /0 B@:0 @:0

Temps get updated in the Temperature graph as well.
Is it possible for PSUcontrol to send a Gcode requesting temp update?

kantlivelong, 2017-04-17:
Interesting. OctoPrint should be sending M105 periodically on its own. You
might want to open a ticket with OctoPrint as that definitely shouldn't
happen.

LundSoftwares, 2017-04-17:
Yes I assume that´s the "Temperature Intervall" setting in Octoprint.
But looking at the terminal now it´s only "Wait" info from the printer, not a single Temp Poll.

I´ll try an ticket at Octoprint intead, thanks alot.

closed 2017-04-18 by kantlivelong

LundSoftwares, 2017-04-19:
Final Comment/Sollution:
I unchecked "Enable automatic Firmware detection" in Octoprint Settings, then checked all options with "Repetier" behind. This made Octoprint send the M105 command at given intervals, and PSUcontrol works lika a charm after that.

kantlivelong, 2017-04-19:
Woo!

 :+1: