kantlivelong/OctoPrint-PSUControl #29: Unable to connect to the printer

Autro, 2017-04-26:
Once I flip the switch on PSU raspberry pi comes to life and loads the system. Meanwhile, everything else is turned off (including printer). Once octoprint is loaded and I click on the button to turn on my ATX PSU printer (Anet A8) turns on and successfully loads the software (I am using Skynet). However, because the printer was not initialized as an USB device originally (that's my guess) I am unable to connect to it. 
Kindly advise on the issue.

kantlivelong, 2017-04-26:
[PortLister](http://plugins.octoprint.org/plugins/portlister/) will automatically connect to your printer once the usb serial device becomes available.

rplc, 2017-05-17:
I just set everything up. If you reload the page the printer is in the menu and you can connect to the printer.

Nethertheless it would be great if the plugin has the option to connect to a printer as soon as the psu-on button is hit.

kantlivelong, 2017-05-17:
@rplc Does **PortLister** not work for you? I just don't see the need to reinvent the wheel if I don't have to.

rplc, 2017-05-17:
On the contrary it works great. I'm just missing an option like connect automatically to the printer as soon as it comes online.

kantlivelong, 2017-05-17:
That's what PortLister does.

rplc, 2017-05-17:
Okay?! Then it is not working properly on my side. For me it just refreshes the serial port Dropdown Box. I then have to click manually on connect.

kantlivelong, 2017-05-17:
Do you have auto-connect enabled?

See
https://github.com/markwal/OctoPrint-PortLister

rplc, 2017-05-17:
My bad...
The autoconnect_delay was set to the standard 20 (seconds). I've never waited 20 seconds...
Just set it to 2 seconds and everything works like a charm.

Sorry for the trouble...

kantlivelong, 2017-05-17:
No worries. :+1:

closed 2017-05-23 by kantlivelong