kantlivelong/OctoPrint-PSUControl #30: Icon botton will not Toggle on or off

shane84948, 2017-05-11:
The icon button is grayed out. I can click it but nothing changes or happens. I'm using Version 1.3.2 (master branch) and PSU control 0.1.4.

kantlivelong, 2017-05-11:
- Could you provide DEBUG logs by following these steps:
- - Enable DEBUG logging ( http://docs.octoprint.org/en/master/configuration/logging_yaml.html ) [Use **octoprint.plugins.psucontrol** for the component name]
- - Stop OctoPrint
- - Clear OctoPrint Logs
- - Start OctoPrint
- - Wait a minute or so.
- - Try switching the PSU on/off (Wait 30 seconds or so in between. Physically switch if needed)
- - Attach logs

shane84948, 2017-05-11:
May sound like a stupid question but where do I find %APPDATA%/OctoPrint.

kantlivelong, 2017-05-11:
That's the Windows path. You need to do this on your OctoPrint device which I am assuming is not Windows as it's not supported by this plugin.

shane84948, 2017-05-11:
I have a raspberry pi zero W which is running Octopi and then Im connecting with octoprint through Crome 
 @ Octopi.local.

kantlivelong, 2017-05-11:
In that case the OctoPrint settings directory is likely located at **/home/pi/.octoprint**

shane84948, 2017-05-11:
When I do it through Octopi i get this

pi@octopi:~ $ ~/.octoprint/config.yaml
-bash: /home/pi/.octoprint/config.yaml: Permission denied

kantlivelong, 2017-05-11:
Basic Linux support is really out of scope here.

Take a screenshot of your configuration for PSU Control.

shane84948, 2017-05-11:
[image]

shane84948, 2017-05-11:
[image]
 
not sure if you can see it but the button is grayed out.

kantlivelong, 2017-05-11:
1. On/Off and Sensing can't use the same GPIO pin.
2. Pin 20 is ground ( https://www.element14.com/community/servlet/JiveServlet/previewBody/80667-102-1-338789/GPIO.png )

shane84948, 2017-05-11:
Thanks and sorry for wasting your time. I was looking at a wrong pin diagram. Now works as advertised.

kantlivelong, 2017-05-11:
No worries. Glad its working :+1:

closed 2017-05-11 by kantlivelong