kantlivelong/OctoPrint-PSUControl #31: Automatic Shutdown after 15 mins idle - waiting for heaters

busaku, 2017-05-11:
Dear PSUControl-Team,

i wish my printer is shutting down, after printing - and then after an idle from 15 minutes.
But the printer e.g. the PSU wont shut down. In the logfiles repeat:

2017-05-11 03:55:00,486 - octoprint.plugins.psucontrol - INFO - Waiting for heaters(tool0) before shutting off PSU...
2017-05-11 03:55:05,498 - octoprint.plugins.psucontrol - INFO - Waiting for heaters(tool0) before shutting off PSU...
2017-05-11 03:55:10,509 - octoprint.plugins.psucontrol - INFO - Waiting for heaters(tool0) before shutting off PSU...
2017-05-11 03:55:15,522 - octoprint.plugins.psucontrol - INFO - Waiting for heaters(tool0) before shutting off PSU...
2017-05-11 03:55:20,533 - octoprint.plugins.psucontrol - INFO - Waiting for heaters(tool0) before shutting off PSU...
2017-05-11 03:55:25,544 - octoprint.plugins.psucontrol - INFO - Waiting for heaters(tool0) before shutting off PSU...
2017-05-11 03:55:30,555 - octoprint.plugins.psucontrol - INFO - Waiting for heaters(tool0) before shutting off PSU...

But it was 3 hours after printing. So, the heaters arent hot anymore. 

Can i disabled waiting for heaters? Or can i setup an force-shutdown after x minutes?

Thanks in advance.

PS: I use the ANET A8 with 2 Port Relai on Raspberry. My Setup from PSU Control is:
GPIO Mode BCM
GPIO Pin for switching (pin 17)
No Sensing
No Power on
Power off is checked, after 15 minutes and 70°
Disconnect on power off is checked.

kantlivelong, 2017-05-11:
Duplicate of #27 ?

busaku, 2017-05-11:
hm... it looks like, thanks :)
i will try, and report my results

busaku, 2017-05-12:
Shut down correctly :)

closed 2017-05-12 by busaku

kantlivelong, 2017-05-12:
@busaku 

Would you be able to provide the output from M115?