kantlivelong/OctoPrint-PSUControl #33: Add togglePSU API Command

kantlivelong, 2017-05-18:


yschroeder, 2017-05-24:
Yes, please! I want to build a keypad that issues API commands and I want to have a button that toggles power. Howevery, I nether found a way via the API to get the current PSU status nor is this toggle option available.

kantlivelong, 2017-05-24:
@yschroeder Yeah status makes sense too.

kantlivelong, 2017-05-24:
@yschroeder I've added both API commands to the **additional_api_commands** branch if you want to test it out and let me know how it goes.

Added:
- togglePSU - Toggles supply on/off. No return data
- getPSUState - Returns isPSUOn value(0/1) as a JSON.

yschroeder, 2017-05-29:
Actually, I have no idea how to use a plugin that is not installed via the integrated GUI. Can you give me a hint on where to find this info?

xoniq, 2017-05-30:
I will try it today. Yesterday mounted a ATX PSU to it, and turning it on and off goes just fine using the lightning icon. Will try it using the API.

xoniq, 2017-05-30:
I'm testing it remotely, don't know I am doing it wrong.
Is this the right way to call it?

`/api/turnPSUOff`

kantlivelong, 2017-05-30:
See #21

yschroeder, 2017-05-30:
I've tested it now. togglePSU works as expected, thanks for that!

However I can't get the `getPSUState` to work. Is it a POST or a GET? Shouldn't it be a GET on `/api/plugin/psucontrol/PSUState` and then return a json object? I tested some variations but can't get the right response back. Please give a hint on how to use this.

kantlivelong, 2017-05-30:
It should be a POST that returns json data but I haven't tested it as I don't have a working environment at the moment.

kantlivelong, 2017-05-30:
curl -s -H "Content-Type: application/json" -H "X-Api-Key: **YOUR_API_KEY**" -X POST -d '{ "command":"getPSUState" }' http://**YOUR_OCTOPRINT_SERVER**/api/plugin/psucontrol

yschroeder, 2017-05-31:
That curl command returns a 500 error.

xoniq, 2017-05-31:
@yschroeder

Remove plugin and add again via the same url. Yesterday they fixed it in a new commit based on my log. (See the other issue)

yschroeder, 2017-06-01:
Getting the status works as described for me after installing again.

I would like to see both features getting deployed to release. For now I will keep using the plugin from the zip file.

closed 2017-06-27 by kantlivelong

gurusonwheels, 2017-08-08:
plugin url does not work . i get 404

kantlivelong, 2017-08-08:
@gurusonwheels its in devel now.