kantlivelong/OctoPrint-PSUControl #34: Add getPSUState API Command

kantlivelong, 2017-05-24:


xoniq, 2017-05-30:
Tried this one command line, got this response:

`curl -s -H "Content-Type: application/json" -H "X-Api-Key: XXXXXXXX" -X POST -d '{ "command":"getPSUState" }' http://XXXXXXXXXXX/api/plugin/psucontrol`

> <!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">
> <title>500 Internal Server Error</title>
> <h1>Internal Server Error</h1>
> <p>The server encountered an internal error and was unable to complete your request.  Either the server is overloaded or there is an error in the application.</p>

The turnPSUOn and turnPSUOff is working however.

kantlivelong, 2017-05-30:
@xoniq could you provide the octoprint log files?

xoniq, 2017-05-30:
The error in the log:

`AttributeError: 'PSUControl' object has no attribute 'isPSUon'`

I did install this one:
https://github.com/kantlivelong/OctoPrint-PSUControl/archive/additional_api_commands.zip
So that should be alright.

Full stack trace:

`2017-05-30 20:34:49,675 - octoprint - ERROR - Exception on /api/plugin/psucontrol [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/server/util/flask.py", line 1106, in decorated_view
    return flask.ext.login.login_required(func)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask_Login-0.2.2-py2.7.egg/flask_login.py", line 637, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/server/api/__init__.py", line 99, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_psucontrol/__init__.py", line 388, in on_api_command
    return jsonify(isPSUOn=int(self.isPSUon))
AttributeError: 'PSUControl' object has no attribute 'isPSUon'
2017-05-30 20:34:49,682 - tornado.access - ERROR - 500 POST /api/plugin/psucontrol (127.0.0.1) 23.53ms`

kantlivelong, 2017-05-30:
Ah I see. Doh!

Can you uninstall and reinstall using the URL then try again?

xoniq, 2017-05-30:
Will do it as soon as my print is finished. Should be within the hour.

xoniq, 2017-05-30:
Success:

```
{
  "isPSUOn": 1
}
```

Thank you for the quick support.

kantlivelong, 2017-05-30:
@xoniq Thanks for testing :+1: 

I've made one additional (and hopefully final) commit to return true/false instead of 0/1 for getPSUState to conform with JSON standards.

closed 2017-06-27 by kantlivelong