kantlivelong/OctoPrint-PSUControl #35: M80 s

broncosis, 2017-06-01:
Marlin has added support for M80 S which gives the status of the power on pin of a board 
this could be used as a alternative to the gpio monitoring

kantlivelong, 2017-06-01:
It looks like this was added less than a month ago and is only available in Marlin. Will revisit when adoption is greater.

broncosis, 2017-06-01:
Yup it's new but figured at least put out there on the radar

⁣Sent from BlueMail ​

aequitas, 2017-07-26:
@broncosis I added a working implementation for this feature, feel free to try it out.

broncosis, 2017-07-26:
I will thank you very much  
I will let you know in a day or 2

⁣Sent from BlueMail ​

spaelectronics, 2017-11-10:
Add support for Smoothie also please!

The command: "M1000 switch psu" will return the following string: "switch psu is 1" or "switch psu is 0" if on or off, respectively.

You can see more info here:
http://smoothieware.org/switch#console-command-to-read-a-switch-state

VjyedSC0UFMc8z, 2021-02-07:
I would like to bump this to support M80 S for sensing as well.

kantlivelong, 2021-04-16:
This can now be done as a sub-plugin if someone desires. Won't be adding to core. See #49

closed 2021-04-16 by kantlivelong