kantlivelong/OctoPrint-PSUControl #36: Invert sense pin

agreen, 2017-06-03:
I hooked up my PSU sense with an optocoupler, but this has a side effect of inverting the logic for the pin. It would be handy if there was an "Invert" checkbox for the sense pin.  I hooked it up using the internal pullup of the rpi GPIO,

agreen, 2017-06-29:
I basically have a PR ready to go for this, however I wanted feedback from you. I have to make one assumption to get it to work right... If you're using an optocoupler and inverting the signal (so that ground is on, and "not ground" is off)... You either need to use an external pullup resistor, or enable internal pullup. What's your thought on enabling internal pullup/down depending on invert sense state? I think it might make the code a bit "safer".. and would still work if there was an external pullup/down.

kantlivelong, 2017-06-29:
I'd personally prefer another option next to the invert checkbox as a dropdown to specify NONE, PUD_UP, PUD_DOWN.

- No Internal Pull-Up/Down
- Internal Pull-Up
- Internal Pull-Down

agreen, 2017-06-29:
Will do. I'll get that added.  Spent a few hours last night scratching my head as to why it wasn't working before I realized.  Also, it's something that should probably be documented onscreen or somewhere. i.e. 
Suggested values:
not inverted = internal pull-down
inverted = internal pull-up

kantlivelong, 2017-06-29:
Would likely just update the README.md as it's already a bit dated by now.

kantlivelong, 2017-08-02:
PR Merged

closed 2017-08-02 by kantlivelong