kantlivelong/OctoPrint-PSUControl #37: ability to add another psu / control

nowkillkennys, 2017-06-14:
ability to add another psu / control two power supplies wit this plugin? (add extra button)?

kantlivelong, 2017-06-14:
Could you provide a use case for this?

nowkillkennys, 2017-06-14:
I have two psu's one for the heat bed and one for the hotend,board,leds
Definitely heats up a lot faster !

kantlivelong, 2017-06-14:
Couldn't you just wire the ON/OFF pin to both PSUs then use an AND gate to determine if both are on for sensing?

nowkillkennys, 2017-06-14:
i use psu control to trigger a relay (for psu1) and i want todo the same for (psu2) via the 2nd switch

kantlivelong, 2017-06-14:
But what exactly is the use case for having separate toggles? Both PSUs should be on while printing and the controller would toggle switching on the bed itself.

I don't think this is something I would pursue adding unless there is enough demand for it. For now you could simply clone the plugin and replace  **psucontrol** with **psucontrol2**.

nowkillkennys, 2017-06-14:
how?

kantlivelong, 2017-06-14:
how what?

nowkillkennys, 2017-06-14:
how do i clone the plug in

kantlivelong, 2017-06-14:
You'll have to download the source, make the changes, and finally upload it to OctoPrint. 

http://docs.octoprint.org/en/master/plugins/

nowkillkennys, 2017-06-14:
i havent a clue how todo all that oh well back to the drawing board thanks for the fast reply

nowkillkennys, 2017-06-14:
i have no programing history anyone want to make a clone of this?

closed 2017-06-27 by kantlivelong