kantlivelong/OctoPrint-PSUControl #41: PSU Auto Off & Auto Connect issues

Mavyre, 2017-07-04:
Hello,

I'm kinda reopening the discussion of the issue #24.
I don't know if it is a bug or a feature, but my Octoprint won't automatically shut off the printer, even if the timout is reached, and there isn't anything in the terminal (except received "wait"s). The manual shutoff from the button works correctly.

The second point I'd like to discuss is the Auto Connect. I know the Port Listener plugin does the job. But it does it for USB only. I'm using the UART port from a Raspberry to communicate with the printer's motherboard (it's clearly better and fits correctly in the case I printed), and in this case, the PortListener doesn't detect any new port (as the UART of the raspberry stays up even if the PSU is off)...

Hope it'll make that awesome plugin even better!

kantlivelong, 2017-07-05:
If you aren't seeing any M105 commands when printing stops then you likely have a firmware detection issue. See #27 

I would prefer any auto connect solution to be another plugin or an enhancement to OctoPrint as I can see it being used for other scenarios where PSUControl is not required.

Mavyre, 2017-07-05:
I see only one M105 after printing. Is it wrong?

Great, I'll have to create my own auto-connect plugin! 😄

kantlivelong, 2017-07-05:
You should be seeing M105 while OctoPrint is connected to the printer. 

Let me know how you make out. Would be good to have a more robust plugin than PortLister.

Mavyre, 2017-07-05:
Okay for now I'll try to correct this M105 issues

Mavyre, 2017-07-05:
Oh wait, my printer just shut off. M105 doesn't seem to be the issue. Trying on a 15 minutes time frame...

kantlivelong, 2017-07-05:
The plugin will basically wait indefinitely or until hotend temps are below the threshold. Turn on debug logging and see what happens.

Mavyre, 2017-07-05:
Oh I think I get the bug. After a print, the temperature stops updating and the plugin - after the delay - thinks it is too hot to shut off.
I don't know how to solve this M105 issue. Could you please tell me where I can report it? It doesn't seem related to your plugin but to OctoPrint (or OctoPi) itself.

By the way, thank you for your support! :)

Mavyre, 2017-07-05:
Ok I spoke once again too much fast. I own an Anet A8, disabled the auto detection of Firmware, and I ticked everything about the Repetier Firmware. Seems to be working now :)

kantlivelong, 2017-07-05:
Close but not exactly. If using **"Automatically turn PSU OFF when idle"** the plugin will turn the supply off after **"Idle Timeout"** but only if the hotend temperature is below **"Wait For Temperature"**.

The M105 issue is typically due to firmware detection. See #27

kantlivelong, 2017-07-05:
Cool :)

Mavyre, 2017-07-05:
I let you close this issue as it's resolved now.

Thank you for all your support!

kantlivelong, 2017-07-05:
No problem

closed 2017-07-05 by kantlivelong