kantlivelong/OctoPrint-PSUControl #42: Add a second relay signal

leobel96, 2017-07-08:


kantlivelong, 2017-07-08:
It's funny you ask. A few days ago I wanted the same thing and ended up stripping the plugin down and making another simple one just for lights. I'll post it in a few.

kantlivelong, 2017-07-08:
https://github.com/kantlivelong/OctoPrint-LightControl

leobel96, 2017-07-08:
What a coincidence ahahah. Thank you. Will it be integrated with PSU control so that with only a switch you can switch on/off them together? Or you will need to act on them separately? When should it be released? Thank you again :)

kantlivelong, 2017-07-08:
No plans on integrating into PSU Control as it's a bit out of scope. Not really intending on releasing Light Control but I use it on mine without issue. Just install the plugin with the zip.

leobel96, 2017-07-08:
Ok. Bye bye.

closed 2017-07-08 by leobel96