kantlivelong/OctoPrint-PSUControl #44: PSU Control turns off printer on initial power up.

Hypertoken, 2017-07-17:
So, my issue is that when I flip the power switch on the side of printer it powers everything up and after a few seconds, maybe 30, the printer powers back off.  It happens only when this plugin is installed. Whether or not I have the "Automatically Turn On", or the "Invert" options checked.

I'm using a relay to switch the 12v from the PSU to the Mainboard, I have it connected to the (NC) pin as I wan't it to power on with the switch. 

The RPi is powered via a 5v regulator that is wired directly to the PSU so it is always on when the physical switch is on.

I tried it with connecting the Mainboard to the (NO) on the relay, but then the sensing pin is reversed. (I even tried agreen "invert sense" pull request to no avail). And I lose the functionality of my printer turning on with the switch. 

Its almost like the plugin wants to (NO) the relay no matter how its wired.

HELP! I have no idea what I'm dong wrong. 

PS: I've made the modifications to the 5V relay board to run off of RPI 3.3V GPIO pins using  a 1k resistor.

kantlivelong, 2017-07-17:
Can you draw a simple schematic of how you have everything wired and include a screenshot of the configuration?

Hypertoken, 2017-07-17:
I have no idea how to draw a schematic, this is the best I can do.
[image]... .Hard to tell on the GPIO on the RPi,  GPIO pin 1 3.3v Yellow, GPIO pin 2 5v Red, GPIO pin 3 Green to DIN1 , Black GPIO pin 6 Ground
[image]

Hypertoken, 2017-07-17:
PS: I have LED's wired on the other Relay, they work fine. Also I can use gpio commands from the command line and it works as expected there.

kantlivelong, 2017-07-17:
I'm not entirely sure that you have that relay wired correctly based on that diagram. VCC will typically receive 5V from the Pi, GND to ground, and then finally IN1 to likely pin 3 on the Pi (though I'd recommend using a green colored pin from [this](http://www.raspberrypi-spy.co.uk/wp-content/uploads/2012/06/Raspberry-Pi-GPIO-Layout-Model-B-Plus-rotated-2700x900.png) diagram)

The sense pin should be a separate GPIO pin that connects to your switched source (I'm guessing 12V) but you need to ensure it only receives 3.3v max. You can use a voltage divider to achieve this.

Once you get that resolved you may need to set the switching pin as inverted and if so it may be worth trying the latest **devel** branch.

Hypertoken, 2017-07-18:
Ok, so my Mainboard has a 3.3v pin, so I wired that to GPIO 7 on the RPi, when the main board is powered GPIO 7 will not get 3.3v... I've also moved IN1 and IN2 to GPIO 11 and 12 (The Green Ones) as you recommended.

Per Foosel (https://github.com/foosel/OctoPrint/wiki/Controlling-a-relay-board-from-your-RPi) I've powered the Relay as such; 
Remove the 2 pin (JD-VCC to VCC) jumper from the relay b
oard, then connect two pins as follows:
*connect JD-VCC to one of the 5V pins on the Pi (pin 2 or 4 on GPIO header)(this powers the relays)
*connect VCC to RPI 3.3V (pin 1 on GPIO header)(this powers the optocouplers)

I've now changed the settings to reflect this; 
[image]
This issue remains after these changes were made.

kantlivelong, 2017-07-18:
I'm honestly confused as to why the relay board would even need to be modified to use 3.3V as the Pi provides 5V but I suppose it works.

It looks as if the relay uses LOW to switch the supply on so invert the On/Off pin.

As a simple test for the sense pin you can try manually switching the supply on without using GPIO to see if the indicator turns on/off. If that is working then you can go from there.

Hypertoken, 2017-07-18:
The sense pin seems to be functioning properly, as if I remove the connection, the Lightning bolt turns from green to grey.
If I use invert on the On/Off pin then it no longer works. As it try's to send the on when its already on. And off when already off.

kantlivelong, 2017-07-18:
So with it not inverted does it function properly when manually clicking the indicator icon (lightning bolt)?

Hypertoken, 2017-07-18:
Yes, actually everything works as expected... its just the fact that when I switch the power on initially, the printer gets power because the relay is NC.. .but for whatever reason it will then switch to NO and power off. 

So a normal power cycle produces this result; 
Power Off via switch - Everything powers off
Power On via switch - Everything powers on
Wait ~10-30seconds - Printer Powers off
Wait for server to come back online (via remote browser) ~10-30seconds
Power printer back on (via several methods, including PSU Control)

as you can see this adds like a minute to my reboot time... :( 

If I disable the plugin, or give it the wrong GPIO info, that doesn't happen.

Hypertoken, 2017-07-18:
I don't think the invert sense logic will work.. because when on, there is 3.3v and when off there will be 0v. If you invert that... 3.3v means that is off... but that cannot happen, because my switched device is now off providing 0v.

kantlivelong, 2017-07-18:
Ah! Sorry about that! There's so many ways these things can be wired/configured. It sounds like you want the default state to be ON whereas the plugin defaults to OFF. Yeah?

Hypertoken, 2017-07-18:
I'm going to try wiring it to NO see if it works proper now that I've switched the sense pins.

kantlivelong, 2017-07-18:
I can add an option to set the default state but I am not sure how it will play out when configuring the pin. I suspect that there may be a short blip in between which would cause the PCB to reset.

If possible see if you can power the PCB with the same source as the Pi so that it is always on but without the main supply.

Hypertoken, 2017-07-18:
No need... I think now that I have it wired "Correctly" things are working. I've now wired the PSU to the NO pin on the relay, and enabled the "Automatically turn PSU ON" option. Now there is just a slight delay after powering on via switch, the printer now will also power on. I'm ok with that!

However I can see where others will logically want to wire their relay to the NC as this will allow the Printer to get power immediately. You may want to add an option for default state for that reason. But for now my issue is resolved.

kantlivelong, 2017-07-18:
Glad its working!

I've added #48 to see if this is possible.

closed 2017-07-18 by Hypertoken