kantlivelong/OctoPrint-PSUControl #51: Remove Pi(RPi.GPIO) requirement

kantlivelong, 2017-07-20:
Remove the RPi.GPIO requirement but disable features if not a Pi.

kantlivelong, 2017-11-05:
This is now available in the **[remove_rpi_gpio_requirement](https://github.com/kantlivelong/OctoPrint-PSUControl/tree/remove_rpi_gpio_requirement)** branch.

Looking for testers.

JoblyFonty, 2017-11-06:
Works for me. :thumbsup:

kantlivelong, 2017-11-07:
@JoblyFonty If using a device other than a Pi would you be able to tell me if the GPIO  options in the **Switching** and **Sensing** drop downs is removed?

JoblyFonty, 2017-11-13:
No, they are present. But I have got RPi.GPIO installed in the venv even though it does not work. So that would explain that.

kantlivelong, 2017-11-13:
I was just looking at this about 10 minutes ago and I think I see/have fixed the problem. Even though RPi.GPIO throws an import error on non-Pi devices it will still be listed in sys.modules. I tested it out here and it seems okay. Mind trying with the latest?

kantlivelong, 2017-11-15:
Merged in devel

closed 2017-11-15 by kantlivelong