kantlivelong/OctoPrint-PSUControl #53: Connecting to printer after power on

SMundt, 2017-07-26:
Is there any possibility to connect to the printer after I've powered it on by the plugin?
Now I must click on the "Connect" button when I powered it on via plugin.

kantlivelong, 2017-07-27:
This may help:

https://github.com/markwal/OctoPrint-PortLister

jneilliii, 2017-07-28:
I would appreciate this option as well.  Similar to the way that you have the option for auto disconnect on power off, it makes sense to be able to control the connection both ways. Maybe with an optional delay for after power off.

kantlivelong, 2017-07-28:
@SMundt @jneilliii 

Could you tell me how you are using this plugin? What features do you currently utilize? How is everything wired?

jneilliii, 2017-07-28:
I'm actually using it with a TP-LINK Smart Plug and this repository https://github.com/softScheck/tplink-smartplug to run system commands to control the plug. I'm using just to be able to turn the printer control box of my Creality CR-10 on and off remotely.

kantlivelong, 2017-07-30:
Hi @jneilliii,

Are you essentially just using the indicator icon to toggle the printer PCB on/off?

I ask because this plugin was(and sort of is still) intended to be used to switch on an auxiliary power source. For example in my case I have an ATX supply that powers my Pi & PCB with the +5VSB that is always-on. Then this plugin simply switches the supply on from standby which provides power for my stepper motors, hotend, and fans.

The way you are using it is still fine but you would need the help of the PortLister plugin to auto connect to your PCB. Additionally even if I was to write some sort of auto connect similar to PortLister I would end up doing it as a separate plugin anyway as it has many other use cases than scope of this plugin.

jneilliii, 2017-07-30:
That's exactly what I'm using it for, just s simple remote power control for the printers pcb. Thanks for the recommendation. I might actually fork your code and simplify it for my needs, as I see your point in regards to it not being designed specifically for the purpose I'm using it for.

kantlivelong, 2017-07-30:
See my LightControl plugin. It's essentially a stripped down version of PSUControl.

You would need to re-add the system command support but shouldn't be much effort.

jneilliii, 2017-09-03:
Just wanted to let you know that I finally got around to programming my version of this plugin, specifically for the TP-Link Smartplug and just wanted to give you thanks for your initial work as mine relies heavily on its core functionality.  You can find it in my [OctoPrint-TPLinkSmartplug](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug) repository.

kantlivelong, 2017-09-06:
Going to close this. May add an option in the future for command based sensing to support devices like this .

closed 2017-09-06 by kantlivelong

kantlivelong, 2017-09-17:
@jneilliii FYI I've added support for sensing via systemcommands in the devel branch with commit 20228cb89f5b689dda5e78abc538037f9a0376f6. This would allow you to use PSUControl in conjunction with a shell script to handle on/off/status.