kantlivelong/OctoPrint-PSUControl #54: [Request] Turn PSU on when OctoPrint server starts up.

aviatorken89, 2017-08-06:
Hello, I love your plugin, but I am using it with a solid state relay to turn the entire printer on and off, so I would like a feature where you click a checkbox, and whenever OctoPrint finished starting up, to turn the PSU on automatically.

Thank you! :)

kantlivelong, 2017-08-06:
You can likely use an event trigger in OctoPrint to call the API and have
the PSU turn on. Search the issues here on github for the appropriate curl
commands.

aviatorken89, 2017-08-06:
I don't see how this is a solution, I would need to put the curl command with an ever changing API key somewhere (I build devel branches of OctoPrint frequently) which means I need to modify the OctoPrint source when all I am asking is put it as a setting in PSUControl. Or if you want me to do it and want me to do a pull request let me know. :)

I forked PSUControl but have done no mods, and didn't really plan on it since the plugin works great.

kantlivelong, 2017-08-06:
Your API key shouldn't change just from changing the binaries. Read it in from the config and use it as a variable. OctoPrint code changes are also not required.

http://docs.octoprint.org/en/master/events/#server

May add as a feature at some point.

kantlivelong, 2017-09-06:
FYI the devel version adds support for pseudo on/off gcode. If not using gcode based switching you can enable it and simply send M80 in the octoprint startup script.

klcjr89, 2017-09-06:
@kantlivelong where is the script located? I can't find it

kantlivelong, 2017-09-06:
Well shoot never-mind I'm dumb. Doesn't apply to this scenario. :frowning:

klcjr89, 2017-09-06:
For now I have just modified my fork as I have not had time to implement the 'turn on psu at server boot' UI yet. It should literally take half hour or less to add a checkbox to the settings UI and add a variable or something.

klcjr89, 2017-09-07:
I have implemented this and have updated the plugin version to `0.1.5` Feel free to use and make any changes, I changed the wording slightly for the UI.

I have tested to make sure it works with rebooting the Raspi and also rebooting just the server; it works a treat. If there's any issue let me know but I would be pleased if you pulled this into the main repo.

Link to PR: https://github.com/kantlivelong/OctoPrint-PSUControl/pull/59

kantlivelong, 2017-09-07:
Thanks for the PR but I think I may implement this in a different way. Something like **PSU State On OctoPrint Start** with a dropdown containing the options "_No Change_", "On", "Off".

Not 100% sure yet.

klcjr89, 2017-09-08:
Thanks, I thought I implemented it in a way that reflects the current design patterns of the plugin including the text PSU ON. It should be simpler to have the one checkbox instead of a dropdown that accomplishes the same thing.

kantlivelong, 2017-09-08:
What you submitted is a valid solution but I just want to take it one step further. My reasoning behind a dropdown with 3 options is that it covers all possible choices..

1. **"No Change"** - Uses existing behavior which does nothing and leaves it at the existing state.
2. **On** - Force On
3. **Off** - Force Off

klcjr89, 2017-09-09:
Ok, if I understand correctly, the reason for the 3rd choice is that if the PSU is on at server boot, that 'Force Off' option will turn the PSU off at server boot, correct?

If so I can see why that makes sense for some people's setups, unless the server crashes during a print the PSU would turn off, but at that point the print is going to be ruined anyways do to the server crashing (I would think?).

kantlivelong, 2017-09-11:
Correct.

Also Correct.

:+1:

Tailslide, 2018-04-03:
Add my vote to this one.. I have a script in my startup that powers on the PSU but then the plugin powers it back off when it starts up.. then I have to manually power it back on again because I can't connect to my board until it gets 12v.   All this powering on and off may have blown a fuse in my board too not sure until more fuses show up in the mail :(

Tailslide, 2018-10-30:
I worked around this by making a couple batch files on my windows client that turns on the PSU before printing.. I have it called automatically from simplify 3d before it sends the file to octopi, then I have it set to automatically turn off after inactivity.  I have a smart power strip with my 24v bed power supply, duct fan, and enclosure heater all turn on when the 12v PSU is triggered. You will need to download curl.exe to c:\unix2 folder to use this. The first script checks to see if it's on, and if not calls the second script that turns it on and waits until it is booted.

**Script1: Print3d.bat**
```

erase \unix2\printeron.txt
C:\unix2\curl -k -H "Content-Type: application/json" -H "X-Api-Key: XXXXXXXXXXXXXXXXXXXXXXXXXXX" -X POST -d "{\"command\":\"getPSUState\"}" http://octopi.mydomain.com/api/plugin/psucontrol > \unix2\printeron.txt

>nul find "false" \unix2\printeron.txt && (
  echo PRINTER IS OFF, TURNING ON
 \batchfiles\printeron.bat
 IF %ERRORLEVEL% NEQ 0 (
   ECHO ERROR STARTING PRINTER
   EXIT /B 1
 )
) || (
  echo PRINTER WAS ALREADY ON
)
echo PRINTER IS ON
```

**Script2: printeron.bat**
 ```
C:\unix2\curl -k -H "Content-Type: application/json" -H "X-Api-Key: XXXXXXXXXXXXXXXXXXXXXXXX" -X POST -d "{\"command\":\"turnPSUOn\"}" http://octopi.mydomain.com/api/plugin/psucontrol > \unix2\printeron.txt

echo CONNECTING
erase \unix2\connect.txt
 C:\unix2\curl -k -H "Content-Type: application/json" -H "X-Api-Key: XXXXXXXXXXXXXXXXXXXXXXXX" -X POST -d "{\"command\":\"connect\"}" http://octopi.mydomain.com/api/connection > \unix2\connect.txt

erase \unix2\printerstatus.txt

:LOOP
echo GETTING STATUS
C:\unix2\curl -k -H "Content-Type: application/json" -H "X-Api-Key: XXXXXXXXXXXXXXXXXXXXXXXX" -X GET http://octopi.mydomain.com/api/printer > \unix2\printerstatus.txt

>nul find "true" \unix2\printerstatus.txt && (
 echo PRINTER CONNECTED
 GOTO DONE
 )

 echo PRINTER CONNECTING WAITING

GOTO LOOP

:DONE

```

bladeslapper, 2021-05-03:
I would love to see this option as well. Until recently, I had an effective work-around by using the enclosure plugin to turn the relay on during server boot but after a recent update of PSUcontrol, the relay turns on and then immediately off again. I have messed with all the PSUcontrol settings to find a solution but disabling seems to be the only thing that works.

> 	Line 33: 2021-05-03 08:05:43,352 - octoprint.server - INFO - Added new permission from plugin psucontrol: PLUGIN_PSUCONTROL_CONTROL (needs: "Need(method='role', value='plugin_psucontrol_admin')")
	Line 33: 2021-05-03 08:05:43,352 - octoprint.server - INFO - Added new permission from plugin psucontrol: PLUGIN_PSUCONTROL_CONTROL (needs: "Need(method='role', value='plugin_psucontrol_admin')")
	Line 33: 2021-05-03 08:05:43,352 - octoprint.server - INFO - Added new permission from plugin psucontrol: PLUGIN_PSUCONTROL_CONTROL (needs: "Need(method='role', value='plugin_psucontrol_admin')")
	Line 68: |  PSU Control (1.0.5) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_psucontrol
	Line 251: 2021-05-03 08:05:55,264 - octoprint.plugins.psucontrol - INFO - Firmware or communication error detected. Turning PSU Off
	Line 252: 2021-05-03 08:05:55,276 - octoprint.plugins.psucontrol - INFO - Switching PSU Off
	Line 253: 2021-05-03 08:05:55,277 - octoprint.plugins.psucontrol - ERROR - Exception while writing GPIO line
	Line 255:   File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_psucontrol/__init__.py", line 529, in turn_psu_off
	Line 274: 2021-05-03 08:05:58,393 - octoprint.plugins.psucontrol - INFO - Periphery version: (2, 3, 0)
	Line 275: 2021-05-03 08:05:58,393 - octoprint.plugins.psucontrol - INFO - Using GPIO for On/Off
	Line 276: 2021-05-03 08:05:58,394 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 23
	Line 281: 2021-05-03 08:05:59,038 - octoprint.plugins.tracking - INFO - Sent tracking event pong, payload: {'version': '1.6.0', 'os': 'linux', 'bits': 32, 'python': '3.7.3', 'pip': '20.3.3', 'cores': 4, 'freq': 1400.0, 'ram': 915718144, 'pi_model': 'Raspberry Pi 3 Model B Plus Rev 1.3', 'octopi_version': '0.18.0', 'plugins': 'toptemp:0.0.1.4,themeify:1.2.2,simpleemergencystop:1.0.5,resource_monitor:0.3.3,psucontrol:1.0.5,printtimegenius:2.2.7,prettygcode:1.2.4,portlister:0.1.10,inlineconfirm:1.1.0,enclosure:4.13.1,dashboard:1.18.2,displaylayerprogress:1.26.0'}