kantlivelong/OctoPrint-PSUControl #55: GPIO pin

joehsmash, 2017-08-23:
Hey, sorry if this has been asked before.

i just installed my ATX psu and its working great so far, how ever i was messing around in the PSU control settings and noticed that there is a way to get it to monitor the state of the PSU. 

currently when i connect to my octoprint it turns the printer on how ever the PSU state is still in the off position.

i understand that i just need to install a jumped from the 3.3v rail to a pin and than define it in the settings.

do i need to download any sort of library to the arduino? or will it just work?

kantlivelong, 2017-08-23:
You can connect the 3.3V from the ATX PSU to one of the GPIO pins on the Pi. I would recommend a decent sized resistor in between.

In my case I chose to use the 12V from the ATX PSU which was then dropped to 3.3V using a voltage divider.

No changes to the printer firmware are required.

joehsmash, 2017-08-23:
So just to clarify, I take my 3,3v add a resistor to it than put it to any pin, and specify in the settings.

Ez pz

kantlivelong, 2017-08-23:
Yep. Any valid GPIO pin.

joehsmash, 2017-08-23:
sorry for my daft-ness 

are we talking on the actual arduino? or can i choose any pin on the ramps?

currently i have an LCD so i cant use any of the digital pins off the back, 

all that's available are Aux 1 (block of 8 pins) and Aux 2 (block of 8 pins) im pretty sure everything else on the board is spoken for in the firmware.

joehsmash, 2017-08-23:
So i created my jumper with a resistor in it, hooked it up to the board, i tried pins D42,D44,D4 & D5. none of them want to work. 

i have also enabled BCM thinking that would make a difference and it didnt. clearly im doing something wrong.

These are the schematics that im using;

https://i.stack.imgur.com/TRInv.png

http://reprap.org/mediawiki/images/3/3f/Arduinomegapololushieldschematic.png

kantlivelong, 2017-08-23:
The 3.3v goes to the Pi GPIO

joehsmash, 2017-08-23:
Wow... I'm something special, 

Carry on nothing to see here

kantlivelong, 2017-08-27:
Assuming this is resolved. Reopen if required.

closed 2017-08-27 by kantlivelong