kantlivelong/OctoPrint-PSUControl #57: Stops working arbitrarily

joseslo, 2017-08-27:
I'm not sure if it's an issue of this plugin or an Octoprint issue. I've installed Octoprint in a Raspberry Pi 2 B+ with this plugin. After several hours running (I use to turn it on in the morning, and the issue usually occurs at night) the Octoprint system menu suddenly disappears. I don't know why but, when this happens, PSU control button (in the navbar) still is there but has no effect. I try to turn the printer off, the confirmation dialog appears normally, I confirm, but nothing happens, and the icon keeps in green, like nothing happened. If I go to a shell and set manually the GPIO pin, it works properly.

kantlivelong, 2017-08-27:
Hi @joseslo,

I haven't seen or heard of this issue occurring. Could you provide log files?

joseslo, 2017-08-28:
The issue regarding the system menu is similar to  [https://github.com/foosel/OctoPrint/issues/797](url), but the proposed solution seems not to work for me.

Here is the log file

`2017-08-27 14:39:05,466 - octoprint.filemanager.analysis - INFO - Starting analysis of local:AAE_pie_endstop_z.gcode
2017-08-27 14:41:21,854 - octoprint.filemanager.analysis - INFO - Analysis of entry local:AAE_pie_endstop_z.gcode finished, needed 136.39s
2017-08-27 14:50:00,352 - octoprint.server - INFO - ******************************************************************************
2017-08-27 14:50:00,357 - octoprint.server - INFO - Starting OctoPrint 1.3.4 (master branch)
2017-08-27 14:50:00,360 - octoprint.server - INFO - ******************************************************************************
2017-08-27 14:50:00,814 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2017-08-27 14:50:10,451 - octoprint.plugin.core - INFO - Found 17 plugin(s) providing 17 mixin implementations, 17 hook handlers
2017-08-27 14:50:10,651 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2017-08-27 14:50:12,443 - octoprint.plugin.core - INFO - Found 17 plugin(s) providing 17 mixin implementations, 17 hook handlers
2017-08-27 14:50:12,737 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2017-08-27 14:50:32,505 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2017-08-27 14:50:33,579 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2017-08-27 14:50:43,167 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python -m pip" as command to invoke pip
2017-08-27 14:50:51,948 - octoprint.util.pip - INFO - Version of pip is 8.1.1
2017-08-27 14:50:51,952 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages, --user flag needed => no, virtual env => yes
2017-08-27 14:50:51,973 - octoprint.plugin.core - INFO - Initialized 17 plugin implementation(s)
2017-08-27 14:50:52,001 - octoprint.plugins.actioncommands - INFO - Add command definition 'action:' = 
2017-08-27 14:50:52,030 - py.warnings - WARNING - /home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/util/__init__.py:591: UnicodeWarning: Unicode unequal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
  if value_source != value_target:

2017-08-27 14:50:52,130 - octoprint.plugins.psucontrol - INFO - Running RPi.GPIO version 0.6.3
2017-08-27 14:50:52,134 - octoprint.plugins.psucontrol - INFO - Using GPIO for On/Off
2017-08-27 14:50:52,137 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 26
2017-08-27 14:50:52,170 - octoprint.plugin.core - INFO - 17 plugin(s) registered with the system:
|  Action Commands (0.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_actioncommands
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/announcements
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/corewizard
|  Cost Plugin (0.1.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_cost
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/cura
|  Custom Control Editor (0.2.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_customControl
|  Detailed Progress Plugin (0.1.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_detailedprogress
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/discovery
|  Enclosure Plugin (3.51) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_enclosure
|  LED Strip Control (0.2.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_LEDStripControl
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/pluginmanager
|  PSU Control (0.1.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_psucontrol
|  Pushbullet (0.1.9) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octobullet
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/softwareupdate
|  STL Viewer (0.3.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stlviewer
|  TouchUI (0.3.8) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_touchui
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/virtual_printer
2017-08-27 14:50:52,184 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2017-08-27 14:50:52,252 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2017-08-27 14:50:52,457 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
[... cut ...]
2017-08-28 06:25:50,604 - octoprint.plugins.announcements - INFO - Loaded channel _octopi from http://octoprint.org/feeds/octopi.xml in 0.47s
2017-08-28 06:27:07,116 - octoprint.server.preemptive_cache - INFO - ... done in 101.28s
2017-08-28 06:27:07,142 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': ''}
2017-08-28 06:27:42,477 - octoprint.server.preemptive_cache - INFO - ... done in 35.34s
2017-08-28 06:34:37,182 - octoprint.plugins.enclosure - INFO - Writing on gpio: 19 value False`

kantlivelong, 2017-08-28:
I see a handful of errors/warning in that log but don't think they would affect the nav bar. Could you try disabling all 3rd party plugins (including this one) and see if you can replicate the problem?

kantlivelong, 2017-09-06:
Closing. Re-open if still an issue.

closed 2017-09-06 by kantlivelong