kantlivelong/OctoPrint-PSUControl #58: Incorrect initial state when GPIO is inverted

anacierdem, 2017-09-06:
I am using this plugin with the invert checkbox checked. This succesfully bring GPIO pin (i.e 18) to low to activate PSU but when I reboot the pi, pin state is low, making PSU run continuously.

When I pick another GPIO pin as a test case the previous pin (18 in this case) stays high, conculding that the plugin is making the transition to the low state.

As this will cause an infinite PSU activation on a power cycle, it should be fixed. I will have a look at the code and send a PR if I have time.

kantlivelong, 2017-09-06:
Try devel first. I believe this is already fixed.

anacierdem, 2017-09-07:
Yes, it may be #40. Will test and inform you on my case.

kantlivelong, 2017-09-18:
This is now in version 0.1.5. Re-open if stil an issue.

closed 2017-09-18 by kantlivelong