kantlivelong/OctoPrint-PSUControl #60: Automatic PSU OFF doesn't work unless connected to server

Jamackey, 2017-09-07:
As soon as I connect back to a server, the Relay switches off.

kantlivelong, 2017-09-07:
- What settings are you using?
- Can you provide DEBUG logs?

Jamackey, 2017-09-08:
Current Settings:
[image]
[image]

How can I find the debug logs?

kantlivelong, 2017-09-11:
Could you provide DEBUG logs by following these steps:
- Enable DEBUG logging ( http://docs.octoprint.org/en/master/configuration/logging_yaml.html ) [Use **octoprint.plugins.psucontrol** for the component name]
- Stop OctoPrint
- Clear OctoPrint Logs
- Start OctoPrint
- Wait a minute or so.
- Recreate issue
- Attach logs

kantlivelong, 2017-09-18:
Hello,

Is this still an issue?

kantlivelong, 2017-10-23:
Closing. Re-open if this is still an issue.

closed 2017-10-23 by kantlivelong