kantlivelong/OctoPrint-PSUControl #61: Set power status to ON when a print starts

noxin1, 2017-09-15:
It's possible for the PSU state to be inconsistent when using something other than GPIO to control the power source.

My use case is I'm using a WiFi outlet and I can turn on the printer manually, but the Plugin still thinks the power is off.   It doesn't impact printing.   It does impact the "Turn off when idle" feature because the Plugin thinks the printer is off already.

kantlivelong, 2017-09-15:
The internal tracking of psu state is something I've been contemplating getting rid of in favor of adding more sensing options (eg Command based sensing so that you can query your plug).

Try using sensing.

kantlivelong, 2017-09-18:
Version 0.1.5 has been released an includes a few options that may be useful for you.

1. You can now use system command based sensing. If your outlet supports returning on/off status you can update your script and use that to get the current state.
2. You can enable psuedo on/off to provide G-Code(M80/M81) functionality when using System Command or GPIO based switching.

Closing this for now but re-open if you have any further questions/comments.

closed 2017-09-18 by kantlivelong

noxin1, 2017-10-06:
Just got around to writing the script and testing this and it worked awesome!   Thanks for taking the time to implement it.

Would it be possible to make the polling interval customizable?    Do you want this as a new issue?

kantlivelong, 2017-10-06:
Hooray! Yeah you can submit it as a FR or make the changes yourself and submit a PR. It's a pretty simple change really.