kantlivelong/OctoPrint-PSUControl #64: Add beforePSUOff & afterPSUOn GCode scripts

kantlivelong, 2017-09-20:
Add a **beforePSUOff** and **afterPSUOn** GCode scripts.

Scripts should be located in {BASEDIR}/scripts/gcode/psucontrol/ and should be editable in similar fashion to the built in **GCODE Scripts**

apprentice10, 2020-06-30:
I'm going to bump this post since I'm running Klipper, After the power is up you should run a "FIRMWARE_RESTART" command to be able to print.

kantlivelong, 2020-12-16:
This is in 0.1.10

closed 2020-12-16 by kantlivelong

CharlieTemplar, 2021-03-06:
> I'm going to bump this post since I'm running Klipper, After the power is up you should run a "FIRMWARE_RESTART" command to be able to print.

Is there anything that klipper needs before powerOff?  and does it need post On delay before FIRMWARE_RESTART?