kantlivelong/OctoPrint-PSUControl #66: Sensing System Command Return Code

tcm0116, 2017-09-25:
I'm having some difficulty getting the Sensing System Command to work correctly. Here's the command I'm using:

`[[ $(python /home/pi/tplink-smartplug/tplink-smartplug.py -t 192.168.1.224 -c info | grep -o 'relay_state.*' | cut -f2 -d: | cut -f1 -d,) -eq 1 ]] || { false; }`

If I run this command from the command line of my RPi followed by `echo $?`, the output is 0 when the power supply is on, and 1 when it is off. However, OctoPrint only reports that the power supply is off when using this command. Am I missing something?

kantlivelong, 2017-09-25:
Could you provide sample output from tplink-smartplug.py for on/off?

tcm0116, 2017-09-26:
Here's the output for the `info` command:

```
Sent:      {"system":{"get_sysinfo":{}}}
Received:  {"system":{"get_sysinfo":{"err_code":0,"sw_ver":"1.1.1 Build 160725 Rel.163650","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"50:C7:BF:53:E6:1E","deviceId":"8006F0F2B5C8FF52B451E5B0AE592C651854AE06","hwId":"5EACBE93FB9E32ECBE1F1C2ADE6DDE11","fwId":"59542398207847F07D67AB57530532B8","oemId":"37589AA1F5CACDC53E2914B7760127E5","alias":"HICTOP Printer","dev_name":"Wi-Fi Smart Plug","icon_hash":"","relay_state":1,"on_time":6290,"active_mode":"schedule","feature":"TIM","updating":0,"rssi":-67,"led_off":0,"latitude":32.911448,"longitude":-97.310440}}}
```

Here's the output for the `on` command:

```
Sent:      {"system":{"set_relay_state":{"state":1}}}
Received:  {"system":{"set_relay_state":{"err_code":0}}}
```

Here's the output for the `off` command:

```
Sent:      {"system":{"set_relay_state":{"state":0}}}
Received:  {"system":{"set_relay_state":{"err_code":0}}}
```

All three set the exit code to 0 on success.

kantlivelong, 2017-09-26:
Alright I think I see what it is. Python on *nix will use **sh** as the interpreter but based on the syntax of your command you want **bash**. Try putting the command into an executable script file and have it specify **bash** as the interpreter or use /path/to/bash -c "YOUR COMMAND"

kantlivelong, 2017-09-28:
Just wanted to follow up with you. Is this resolved? Can I close this?

tcm0116, 2017-09-28:
I haven't had a chance to try your suggestion. I'll give it a try this evening. ​

tcm0116, 2017-09-30:
Putting the command in a bash script does not seems to be working. I determined that the info request need to be sent a few seconds after the command, so I added a sleep in the script to give it time to change states, but that didn't help. I was also having an issue with the `-c` parameter to bash where it wouldn't process my command line arguments to the scrip, but removing that didn't help either.

kantlivelong, 2017-10-01:
There is a postOnDelay setting to wait X seconds after powering on but aside from that the PSU status will be checked every 5 seconds. If it works in a bash script and you have the plugin call said script it _should_ be the same result.

kantlivelong, 2017-10-26:
Closing. Please re-open if necessary.

closed 2017-10-26 by kantlivelong