kantlivelong/OctoPrint-PSUControl #69: Startup javascript error

BillyBlaze, 2017-10-17:
While I was working on your plugin within TouchUI, I noticed the following javascript error during startup (Chrome 61, OctoPrint 1.3.5, PSUControl 0.1.5):

```Error calling onDataUpdaterPluginMessage on view model PSUControlViewModel : TypeError: Cannot read property 'css' of undefined```

It is referring to [this line](https://github.com/kantlivelong/OctoPrint-PSUControl/blob/master/octoprint_psucontrol/static/js/psucontrol.js#L27).

`#psucontrol_indicator` is available from the start, why not store that directly with jquery instead of undefined?

kantlivelong, 2017-10-17:
See https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/151

BillyBlaze, 2017-10-17:
Yes, I know, but the bug is on your end, I cannot fix this from TouchUI.

Currently the method onDataUpdaterPluginMessage gets called before onAfterBinding, so your property `self.psu_indicator` is still undefined and not a jquery object.

As previously suggested, I suggest changing your code to

```
        self.psu_indicator = $("#psucontrol_indicator");

        self.onAfterBinding = function() {
            self.settings = self.global_settings.settings.plugins.psucontrol;
        };

        self.onDataUpdaterPluginMessage = function(plugin, data) {
            if (plugin != "psucontrol") {
                return;
            }

            self.isPSUOn(data.isPSUOn);

            if (self.isPSUOn()) {
                self.psu_indicator.css('color', '#00FF00');
            } else {
                self.psu_indicator.css('color', '#808080');
            }

        };
```

This will fix your issue. I have tested it myself 👍

kantlivelong, 2017-10-17:
Oh its you! Didn't recognize in my mail client.

I’ll take a look in a few.

kantlivelong, 2017-10-17:
Odd. I just tested both before and after the change. Neither result in a js error and both work for toggling but no indicator is visible.

BillyBlaze, 2017-10-17:
Hmm very odd, dunno how TouchUI would influence OctoPrint's binding calls. Anyhow, I would still use the latter code to be sure it wont be thrown again :)

btw, after some tweaking this is how your plugin will be represented in 0.3.9 (releasing in a few hours)
[image]

kantlivelong, 2017-10-17:
Agreed. 

Just tested your maintenance branch and it looks good on my end. Thanks! :+1:

kantlivelong, 2017-11-05:
Fixed in **devel**. Thanks :+1:

closed 2017-11-05 by kantlivelong