kantlivelong/OctoPrint-PSUControl #70: Hello.

MarioP73, 2017-10-20:
I installed the "PSU Control" plugin and also included a clone of which is called "LED Control" and the lighting switches. Works perfect. I would like to have a different symbol in the head to control, since two flashes are very confusing. But I could not find any reference to the symbol in your code.

How can I include a different symbol?

[image]
[image]

Mario

kantlivelong, 2017-10-20:
Fork the plugin and modify it. Alternatively if you are simply turning the LEDs on/off see https://github.com/kantlivelong/OctoPrint-LightControl. It uses a lightbulb indicator icon.

MarioP73, 2017-10-20:
The LightControl has unfortunately not the functions like the PSU, however a nice symbol. The Invert of the GPIO unfortunately does not work with the LightControl. Both presets result in the same result. Would a full version of the LightControl like the PSU possible?

Mario
[image]

kantlivelong, 2017-10-20:
I have no plans to make any changes to LightControl. I made it for my own use and shared it if other want it. In my opinion I would ask the dev of LEDControl to use the lightbulb icon.

MarioP73, 2017-10-20:
LED control is actually by you. I have adapted the code so that a new plugin has emerged. This allowed me to install PSU and LED in parallel. The references in the LED Control are even yours. The changes I made by an editor, which has replaced all characters with "PSU" and "psu" against "LED" / "led".

Mario

kantlivelong, 2017-10-20:
I noticed as I typed that. You should be able to fork LightControl and add the invert option fairly easily.

closed 2017-10-23 by kantlivelong