kantlivelong/OctoPrint-PSUControl #71: Create TPLink Smart Plug helper script

kantlivelong, 2017-10-24:
Simply python script that can be called for on/off/sensing using system commands.

kantlivelong, 2017-10-24:
Work in progress at https://github.com/kantlivelong/OctoPrint-PSUControl/blob/tplink/scripts/TPLinkSmartPlug/psucontrol_tplinksmartplug.py

Bazeone, 2017-11-24:
where would I place this file to test it out?

kantlivelong, 2017-11-25:
Anywhere on the filesystem where OctoPrint has access. Keep in mind that I may not go forward with this as is.

kantlivelong, 2017-12-20:
I've decided to abandon the idea of supporting this device myself as I do not personally own one and have no need for it.

If anyone wants to write a howto that would be great.

[This](https://github.com/GadgetReactor/pyHS100) library/utility seemed interesting.

jneilliii, 2017-12-24:
You could potentially use my plugin...

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug

kantlivelong, 2017-12-26:
Hey @jneilliii. The TPLinkSmartPlug plugin would certainly work for users who wish to use it. My goal with PSU Control is to keep it generic and support any hardware.

jneilliii, 2017-12-26:
For those of you that do want this feature with PSU Control, that is how I initially was doing it using system commands. To achieve this here are the steps.

ssh to your pi and run the following command to download the script:
`wget https://raw.githubusercontent.com/softScheck/tplink-smartplug/master/tplink-smartplug.py
chmod +X ./tplink-smartplug.py`

In PSU Control, change switching method to System Command and enter the following replacing <ip of plug> with your plug's ip.
On: /home/pi/tplink-smartplug.py -t <ip of plug> -c on
Off: /home/pi/tplink-smartplug.py -t <ip of plug> -c off

kantlivelong, 2017-12-26:
Thanks @jneilliii. You wouldn't happen to have a sample output from checking on/off state would you?

jneilliii, 2017-12-26:
If you run the following command:

`tplink-smartplug.py -t 192.168.1.130 -c info`

the output comes out like this:

```
Sent:      {"system":{"get_sysinfo":{}}}
Received:  {"system":{"get_sysinfo":{"err_code":0,"sw_ver":"1.1.2 Build 161014 Rel.164426","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"50:C7:BF:6A:6C:83","deviceId":"8006DB119EADD41BCE508828B19A949F1872F15C","hwId":"5EACBE93FB9E32ECBE1F1C2ADE6DDE11","fwId":"6C4ACA62CB1B921060CF24148345CD43","oemId":"37589AA1F5CACDC53E2914B7760127E5","alias":"#","dev_name":"Wi-Fi Smart Plug","icon_hash":"","relay_state":0,"on_time":0,"active_mode":"schedule","feature":"TIM","updating":0,"rssi":-72,"led_off":0,"latitude"#,"longitude":#}}}
```

what you key off of for current state is relay_state in that.  If it's 0 it's off if it's 1 it's on.

kantlivelong, 2017-12-27:
So something like this would work?
> (exit $(tplink-smartplug.py -t <ip of plug> -c info | sed 's/,/\n/g' | grep 'relay_state' | awk -F':' ' { print $2 } '))

jneilliii, 2017-12-27:
I just tested that command exactly filling in my plugs ip address and it didn't return anything.  I ended up removing your exit part and all the parenthesis and it returns either 0 or 1 depending on plug state as expected.  Here's what I ended up running in ssh to get it to return anything.

`/home/pi/tplink-smartplug.py -t 192.168.1.130 -c info | sed 's/,/\n/g' | grep 'relay_state' | awk -F':' ' { print $2 } '`

kantlivelong, 2017-12-27:
The exit part returns the on/off status via the exit code. After running what I supplied do:
> echo $?

jneilliii, 2017-12-27:
Yes, that seems to echo the 0 or 1 depending on current plug state.

jneilliii, 2017-12-27:
But putting that in the sensing section using system command doesn't seem to work to adjust the proper current state.  I assume that would have to be adapted for some other way. Let me know if there's anything else you'd like me to test.

kantlivelong, 2017-12-27:
If you're interested in testing [this](https://gist.github.com/2d9b4d8e24569e7f18969214d7ec75c7) bash script it would be great. Place in the same directory as the py or edit to specify the appropriate path. (Don't forget to set the execute bit)

Use the following settings in PSU Control:
On Command: /path/to/bash_script.sh PLUGIP on
Off Command: /path/to/bash_script.sh PLUGIP off
Sense Command: /path/to/bash_script.sh PLUGIP sense

Thanks!

jneilliii, 2017-12-27:
After changing the path in the bash script to full path (even though it was in same directory as the py file) it seems to work for turning on and off the plug, but not sensing the current state. Running the script in ssh seems to do the same thing as before and doesn't return anything.

kantlivelong, 2017-12-27:
Ah my little exit trick neglected to invert the value. Always fun doing this without hardware. Here's the latest with some additional output to make things easier.

https://gist.github.com/45ed9ae8bb834b3bdddf1430f151889c

kantlivelong, 2017-12-27:
https://gist.github.com/75294394a8d330e530a0d16a50acbe7a

kantlivelong, 2017-12-27:
Ugh sorry about the spam. Just comment out/fix the SCRIPT_PATH before running.

jneilliii, 2017-12-27:
Yep. that seems to work after removing the second script path and adjusting the first one to the full path of the py file.

kantlivelong, 2017-12-27:
Thanks @jneilliii.

I'll commit it and write a simple doc.

kantlivelong, 2017-12-27:
Guide complete.

https://github.com/kantlivelong/OctoPrint-PSUControl/wiki/Scripts#tplink-smart-plug

closed 2017-12-27 by kantlivelong