kantlivelong/OctoPrint-PSUControl #72: PSU not switched automatically

richardvdweerd, 2017-11-05:
Hello,

I'm using PSUControl for a while now, but never to get it shutting down or starting up the PSU.

I thought that powering ON was working okay, but I had a M80 in the 'Before print job starts' of Octoprint Setting. When I removed that the printer was not powering on anymore. The flash-icon turned green correctly when starting a print from Cura, but the power didn't switched on.
When issuing a M80 or M81 from Octoprints Terminal, or clicking the flash-icon, the power is turned on/off correctly.
My PSU-relay is connected to my Megatronics, not to the Rpi. 

Another issue: My Sensing-Method is 'Internal', but after resetting/restarting, that setting is gone. 
BTW, is that the correct setting?

My PSUControl settings:

[image]

What's going wrong? Do I use wrong settings?

Thanks
Richard

kantlivelong, 2017-11-05:
Just to confirm, OctoPrint is connected to the controller and the PSU is off. You send M80 to the terminal and it does or does not turn on?

As for Sensing you should really use **GPIO** or **System Command**. Internal is very basic.

richardvdweerd, 2017-11-06:
Switched the relay from the Megatronics to the Rpi. Switching on and (timed out) switching off works fine now.

Sensing is still on 'internal', otherwise timed out switch off does not work.

Thanks!

closed 2017-11-06 by richardvdweerd