kantlivelong/OctoPrint-PSUControl #74: HTTP relay control

Mungbeanz, 2017-11-09:
I have a Sonoff S20 flashed with Espurna this works well using http command:
http://192.168.1.100/api/relay/0?apikey=1234&value=1
in any web interface 

or a python script:
```
import requests
r = requests.get('http://192.168.1.100/api/relay/0?apikey=1234&value=1')
print r.text
```

I tried calling on the Switching.
Switching method (System Command)
On system Command: /home/pi/scripts/PowerScripts/python PowerOn.py
and also just using the url (http://192.168.1.100/api/relay/0?apikey=1234&value=1)

Is there something I am missing.

I looked into the TP link script and how that was run but that seems to have more functions built in.

Thanks,

Dan

kantlivelong, 2017-11-09:
Assuming the python script worked I see a space in the script name. Either
escape it, put in quotes, or replace the spaces with underscores. You may
also want to prefix the command line with python since the script doesnt
specify the python interpreter and I am not sure if it has exec perms.

Mungbeanz, 2017-11-09:
Wow, you are awesome.  It worked.

On System Command:  python /home/pi/scripts/PowerScripts/PowerOn.py
Off System Command:  python /home/pi/scripts/PowerScripts/PowerOff.py

kantlivelong, 2017-11-09:
Woo! For Sensing the script should exit with a 0 for on and 1 for off.

Mungbeanz, 2017-11-09:
Was just going to use Sonoff S20 for turning off printer 5 mins after completing a print.  I will turn it on by the button on the plug.  I was having an issue with octoprint not being able to connect to the printer if the printer is turned on after octoprint is booted.  This way I can turn printer on by button, then boot octoprint, and have it safely turn off when print is finished.

I will give this a go after the print is finished, but my test with the lamp is working well so far.

Any other reason you think I would need to work on sensing? or just leave it disabled?

kantlivelong, 2017-11-09:
I generally just recommend using either command or gpio based sensing as it
is more reliable way of tracking the state. Internal is more of a “getting
started” option.

kantlivelong, 2017-11-09:
If you end up writing a formal script I'd be happy to incorporate into the Wiki for others.

Other than that, are we all set to close this?

Mungbeanz, 2017-11-11:
Yeah, all solved.  Ill do a writeup later on to help others if needed.

Did a full test print worked perfectly.

Thanks,

kantlivelong, 2017-11-13:
Great!

closed 2017-11-13 by kantlivelong