kantlivelong/OctoPrint-PSUControl #76: Gpio pins only work once

siblues, 2017-11-18:
I don't know if I am doing something wrong or is this is how it was made to work but my Psu control will only trigger the relay once in either direction either it closes once or opens once.That is happening on every gpio I select.The only things on my RPi3 are Octopi-Jessie lite and drivers for the 3.5 TFT.I followed all the wiki's on here on how to wire and set everything up but now I am kind of stuck.I have absolutely no experience at this and have gotten this far.I am sorry if this isnt the right place to bring this up.

kantlivelong, 2017-11-18:
Can you provide a screenshot of your config as well as a schematic for how
the relay is wired.

siblues, 2017-11-18:
Please ecxuse my simplistic understanding of all of this but do you mean a screenshot of the config.txt or of how I have the PSU Control settings in Octoprint?I am using the how to on the Foosel/Octoprint page at

[]https://github.com/foosel/OctoPrint/wiki/Controlling-a-relay-board-from-your-RPi(url)

kantlivelong, 2017-11-18:
Screenshot of the settings for PSUControl as well as a drawing of how you
have the relay wired.

siblues, 2017-11-18:
Here is the screenshot 

[image]

Here is my horrible paint wiring diagram
[image]

kantlivelong, 2017-11-18:
1. What model/part number is the relay?
2. What is GPIO 24 used for?
3. What is the relay switching?

siblues, 2017-11-18:
I ordered the one linked in the thread it is the Sainsmart 4 channel relay sku 101-70-101.

Gpio is used to turn on and off the printer it is listed as number 18 in the PSU control and they both work they turn off or on once but then thats it they won't work again unless I change the settings inside of octoprint to control another GPIO and then that one will work the first time but neither will work more than once.The relay is switching led's and the printer power supply.I am not using the mains just the 12v coming out of the supply.

rene3d, 2017-11-18:
Hello

i have the same setup an the same issue. 
The relay is allways on, i can not switch, with the icon in the navbar.

kantlivelong, 2017-11-18:
It looks as if the mentioned relay module uses active-low signals. On a Raspberry Pi you will need to use a transistor to tie it to ground when on. As a simple test you should be able to switch a pin on and off by touching it to ground.

rene3d, 2017-11-18:
Hi 

i found my failure, the VCC was 5V not 3,3V, wrong Pin. Everthing is fine.

kantlivelong, 2017-11-18:
@rene3d Are you using the same relay module?

siblues, 2017-11-20:
This can be closed after redoing the full install a few times it just started working.

closed 2017-11-20 by kantlivelong