kantlivelong/OctoPrint-PSUControl #77: GPIO to turn on PSU

DkOuvE, 2017-11-19:
It would be great for there to be a feature where if there is a pusle on a GPIO pin, the PSU is turned on.

I've got a momentary push button and it would be better to be able to press it and have the PSU turn on than to get up OctoPrint and toggle it that way. There could also be a long press to turn it off if needed.

kantlivelong, 2017-11-19:
Not exactly sure what you are asking but perhaps it's related to #28?

DkOuvE, 2017-11-19:
Oh I didn't know that it was already in the works!

kantlivelong, 2017-11-19:
From what I understand it worked but needed some tweaks.

kantlivelong, 2021-04-05:
This is now possible via a plugin as of 27e66a21e91cfb855bf1b78006c16cbd0cedd632 in devel. Still no plans for adding in core.