kantlivelong/OctoPrint-PSUControl #79: [Request] small change for using it on non Raspberry Pi devices

gdachs, 2017-12-05:
Would you please be so kind and allow to use your plugin without the RPi.GPIO module?
You could wrap the import between try: and except:. In the except part you could remove the option for using GPIO somehow. Your plugin would be generally usable.
I use a Odroid C2 that is way more powerful than a RPi. The Odroid C2 has no RPi.GPIO (the Odroid C1 has :( ). It has wiringPI, but my Python knowledge is not good enough for a fork, and I have no time.

Thanks
Gerald

kantlivelong, 2017-12-05:
This has actually been added in the **devel** branch if you would like to give it a try. :smile:

Related to #51

gdachs, 2017-12-05:
Sorry, should have looked for closed issues.
Installs fine. Toggle button shows up in title bar and menu of touchUi. Settings dialog seems to work.
I can't make real tests, as the first parts of my new Hypercube Evolution are not delivered till now.
Thank you very much, now I can write some bash scripts that use the GPIO command from wiringPi :)

closed 2017-12-05 by gdachs

kantlivelong, 2017-12-05:
Could you confirm if GPIO related options are not visible in settings?

gdachs, 2017-12-05:
yes, I see only System and G-Code commands. In sensing are only Internal and system.

I have enabled "Show warning dialog when powering off via toggle button.", but nothing happens if I hit the button. Is it because I currently have nothing else entered in the settings?

kantlivelong, 2017-12-05:
Perfect. Thanks

You will need to configure switching and sensing before the toggle will work. You can use internal sensing to start off with but definitely move to system command when ready. As a dumb test you could set the switching on/off commands to "true". 

https://github.com/kantlivelong/OctoPrint-PSUControl/wiki/Settings

gdachs, 2017-12-05:
It work!