kantlivelong/OctoPrint-PSUControl #81: Needs a wiring diagram

targetdrone, 2017-12-11:
I'm sorry if it's obvious, but I'm a newb to connecting hardware to the pi.  My pi is powered by a wall wart, independent of the ATX power supply.  The wall wart has no ground pin.  Because I have a 12" build plate, I blew out my 29A 12VDC power supply, so I bought a 750W ATX power supply, and a small circuit board to run it.  I now have plenty of power.  The PSU has a ground pin, of course.

Is it safe to just connect pin 40 from the GPIO directly to the sense pin on the ATX cable?  Do I also need to interconnect the ground wires of each, or does that risk some kind of unsafe interconnection problem if the two grounds are not at the same potential?  If it's a problem, should I be trying to power the pi from the ATX power supply?  I've read that it is designed to be powered via USB only, and not from the 5V rail on the GPIO header.

My hope is that a wiring diagram would answer these questions.

kantlivelong, 2017-12-11:
I'm out of town for a few days but will try to draw up some schematics by end of week.

kantlivelong, 2017-12-16:
Still a work in progress but [this](https://github.com/kantlivelong/OctoPrint-PSUControl/wiki/Wiring-Diagrams) is how I wired mine.

tcj-lev, 2018-04-18:
you can also use the 3V3 from the ATX PSU to trigger PI_GPIO_2 which would be straight-forward and need less soldering etc.

kantlivelong, 2018-04-18:
@tcj-lev Yep that is entirely doable using the PWR_OK pin.

StudioKamar, 2018-07-23:
Can You provide schematic once more because i bought the parts and now schematic not is not schowing

Badn1nja, 2021-04-30:
You can wire into the GPIO pins that are found here; https://www.raspberrypi.org/documentation/usage/gpio/
[image]

In order to select the pins for the PSU Control that can be done as below; Here I'm using GPIO Pin2 
[image]

huschpuscheli, 2022-02-07:
sorry, if if this a stupid question but i have a similar setup like @targetdrone: how do i connect the atx psu with my rpi to switch it on? via the atx's 3v3? ps_on? pw_ok? i'm confused by @tcj-lev's and @kantlivelong's statements about the atx's 3v3 and pw_ok pins. there's also the ps_on pin on the atx which i used to short with gnd to switch on the psu but i want my rpi (which is powered by the atx's 5vsb) to do that. is this possible by psucontrol?

huschpuscheli, 2022-02-15:
> @tcj-lev Yep that is entirely doable using the PWR_OK pin.

is it the pw_ok or the ps_on pin? afaik, pw_ok works only after ps_on, right?