kantlivelong/OctoPrint-PSUControl #82: PSU-Control fails from time to time

azalanono, 2017-12-11:
Hello. 
From time to time I notice that PSU control fails either on automatically turning the ATX power on or off.
It happened right now again that it turned the ATX on correctly, but didn't turn it off (but the temperature of extruder was already on room temperature for more than 2 hours. (I came back into the room and noticed that all fans were still running). I clicked then the "flash-icon" in octoprint and PSU control turned the ATX off.

My suggestion is to slightly extend PSU control. It will hurt nobody when PSU-control repeatedly checks the temperature and repeats from time to time the ON and OFF command.  I don't know if the ATX status can be read back from marlin, but sending a G80 or G81 every five minutes should not be  problem at all and ensure that - even with delay - the PSU will finally be switched correctly.

kantlivelong, 2017-12-11:
Going to need more info.

- What hardware is OctoPrint running on?
- OctoPrint Version?
- Firmware Version?
- PSUControl Version?

Additionally please provide DEBUG logs with the issue occurring. Steps below:
- Enable DEBUG logging ( http://docs.octoprint.org/en/master/configuration/logging_yaml.html ) [Use **octoprint.plugins.psucontrol** for the component name]
- Stop OctoPrint
- Clear OctoPrint Logs
- Start OctoPrint
- Replicate the issue
- Attach logs

azalanono, 2017-12-15:
Hi, Bruce.

Hardware is a good old raspberry Pi version "B". The first one. 
Printer is based on Marlin.  Nothing has changed on hardware and firmware within last 3-4 months. Everything worked before your last update (except of 1-2 cases that it didn't turn PSU off, it always worked).

Software:  It is actually "octopi" with all automated updates.
pi@octopi:~ $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@octopi:~

This is one of the latest logs (but not the very latest one).
[octoprint.log.zip](https://github.com/kantlivelong/OctoPrint-PSUControl/files/1562637/octoprint.log.zip)

As I changed right now the folders for recording timelapses to an nfs mount (don't want to write my SD to death with unnecessary writing operations), I have now restarted octoprint. A job is running, I will see what happens afterwards.

azalanono, 2017-12-15:
I forgot to mention that absolutely nothing appears in point of psucontrol tries to turn the PSU off, but i see that it logs that it turns the PSU on.

Suggestion:
it would be a good (safe) approach to send the PSU-ON/OFF command in intervals of some minutes. 
So when one command got lost/ignored/..., it will come again after 2-3 minutes.   Perhaps add a field to PSU Control where I can "re-send ON/OFF command interval" in ... minutes (or seconds)?

kantlivelong, 2017-12-15:
Hi,

The logs that were attached are not fresh and I am not seeing any DEBUG statements. Can you follow the steps outlined and reattach?

Thanks

azalanono, 2017-12-15:
Hi, Bruce.
I am always very careful when it comes to modify octoprint (as I am not familiar with its internals), so I please for confirming that I did it right.

I created the file ~/.octoprint/logging.yaml (owned by pi:pi with right 644) with following content:
```
loggers:
  octoprint.plugin.psucontrol:
    level: DEBUG
```
Then I restarted octoprint from its menu. Will this create the expected debug-level logfile you need?  I cannot start a new printjob right now because I have a part on the printer and I am not there till monday.

This might be of interest for you:

- After a printjob, the PSU remains on until I click then the PSU-Control icon. Only then it turns the PSU off.

- When PSU is off and I enter a command, the PSU is automatically turned on, but now it is automatically turned off when IDLE time is reached.

kantlivelong, 2017-12-15:
Looks correct. 

I would also suggest enabling serial logging as that would help a lot. The only downside is that it can impact performance depending on the print.

> When PSU is off and I enter a command, the PSU is automatically turned on, but now it is automatically turned off when IDLE time is reached.
Are you saying that it works properly when simply running a command(I imagine non-temperature related) and then waiting for timeout?

See if you can replicate the issue with the following tests:

#### Test 1
1. Ensure PSU is OFF
2. Manually send `M104 S0`
3. Wait timeout and see if it shuts off.

#### Test 2
1. Ensure PSU is OFF
2. Manually send `M104 S100`   _Ensure that it is safe to heat your hotend_
3. Wait timeout and see if it shuts off a few minutes after (timeout + cooldown time)

If either test results in the PSU not shutting off then send the logs.

azalanono, 2017-12-15:
Hi, Bruce.
Surprisingly both tests ended with a **turned-off PSU**. As it should.

In Test 2, nozzle heating turns off after IDLE time, and when temperature goes below 50°C, PSU turns off, too.

kantlivelong, 2017-12-15:
Do you know if this occurs with a specific gcode file or any? If a specific gcode file, can you upload the gcode from the print that failed as well the contents of ~/.octoprint/scripts/gcode?

azalanono, 2017-12-18:
Hi, Bruce.
Today (back in my office) I did several tests.
It does not matter what file I send. today it was even worser than before. I tried to repeat what we tested on weekend. I manually turned off the PSU by clicking the flash-icon, then sent "M104 S55", waited that the extruder exceeds 50°C, and finally sent "M104 S0".   This time it did not turn off the PSU.

Here you find the requested scripts. 
[all_scripts.zip](https://github.com/kantlivelong/OctoPrint-PSUControl/files/1567054/all_scripts.zip)

I also attach here the testing G-Code file (downloaded from printer, so it is what actually is stored there).
[P_coin.gcode.zip](https://github.com/kantlivelong/OctoPrint-PSUControl/files/1567059/P_coin.gcode.zip)

kantlivelong, 2017-12-18:
Since you are able to easily replicate it now can you supply fresh DEBUG logs with the issue occurring?

kantlivelong, 2017-12-18:
... as well as serial logs.

kantlivelong, 2017-12-19:
I just noticed. Your logging like has octoprint.plugin.psucontrol instead of octoprint.plugin**s**.psucontrol. Bad on my part for missing when checking but shoot the logs whenever you can.

azalanono, 2017-12-21:
Hi, Bruce.
I also was wondering about how you will find any useful detail in the log (noticed no difference), but now it is definitely verbose. :) 

The behaviour is now as follows:

- PSU starts up with printjob (as exprected).

- After the printjob finished, I can see the decreasing extruder temperature while I watch in parallel the temperature graph and the logfile.
- When the temperature got below 50°C, I'd expect that something would appear in the log, but nothing happens.
- Finally I clicked the "Flash icon" to turn the PSU manually off. In same moment I got this logfile entries: 
```
2017-12-21 15:46:26,661 - octoprint.plugins.psucontrol - INFO - Switching PSU Off
2017-12-21 15:46:26,668 - octoprint.plugins.psucontrol - DEBUG - Switching PSU Off Using GCODE: M81
```

After that I used the octoprint terminal to manually set the temperature to 55°C (M104 S55) and to turn heating off again (M104 S0).
```
2017-12-21 15:48:35,167 - octoprint.plugins.psucontrol - INFO - Auto-On - Turning PSU On (Triggered by M104)
2017-12-21 15:48:35,171 - octoprint.plugins.psucontrol - INFO - Switching PSU On
2
```
THIS TIME it triggered and did shut off the PSU, but this happened with a delay (when timeout triggered), and not when the nozzle temperature got below 50°C.
[octoprint.log.zip](https://github.com/kantlivelong/OctoPrint-PSUControl/files/1579596/octoprint.log.zip)

Seems that PSU control cannot read the temperature?

kantlivelong, 2017-12-21:
Need a serial log.

azalanono, 2017-12-21:
Hi, Bruce.
I did another test print meanwhile. Same again.  
```
2017-12-21 16:26:18,324 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2017-12-21 16:26:23,342 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2017-12-21 16:26:28,357 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2017-12-21 16:26:33,371 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2017-12-21 16:26:35,640 - octoprint.util.comm - INFO - Finished in 846.090 s.
2017-12-21 16:26:35,689 - octoprint.plugins.detailedprogress - INFO - Printing stopped. Detailed progress stopped.
2017-12-21 16:26:38,386 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2017-12-21 16:26:43,464 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2017-12-21 16:26:48,486 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
2017-12-21 16:26:53,586 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: True
```
The message in last line repeats infinitely, even when both (nozzle and bed) are cold.

About serial log...
Please tell me exactly what I have to do to enable it.
```
pi@octopi:~/.octoprint $ cat logging.yaml
loggers:
  octoprint.plugins.psucontrol:
    level: DEBUG

# serial.log
#serialFile:
#  class: logging.handlers.RotatingFileHandler
#    level: DEBUG
#    formatter: simple
#    maxBytes: 2097152 # 2 * 1024 * 1024 = 2 MB in bytes
#    filename: /path/to/octoprints/logs/serial.log
```
When uncommenting last 6 lines, octoprint does not start up any more.

kantlivelong, 2017-12-21:
Going from memory but Settings > Serial > Enable serial logging.

Make sure you send both fresh octoprint.log and serial.log files with the
issue occurring.

azalanono, 2017-12-21:
Please show me that lines I have to copy&paste into ~/.octoprint/logging.yaml

I noticed that the above path is wrong (oversaw that) and tried then /home/pi/.octoprint/logs/serial.log as well as a nfs share (/goliath/pi/serial.log).   In both cases octoprint terminates its startup after a few seconds.

kantlivelong, 2017-12-21:
http://docs.octoprint.org/en/master/configuration/config_yaml.html#id16

ntoff, 2017-12-21:
You can enable serial logging here

[image]

No need to mess with configuration files manually. Once ticked, restart octoprint (I think, do it anyway just to be safe) and then the serial.log will appear in the regular logs section of octoprint's settings where you can download all the other logs.

Just, whatever you do, do **NOT** post any configuration files publicly, especially config.yaml, as it can contain things like your API key which can be used to gain full control over your printer if it's exposed to the internet.

azalanono, 2017-12-21:
Ok, here with serial log, too....
[logs.zip](https://github.com/kantlivelong/OctoPrint-PSUControl/files/1579928/logs.zip)

And here some time marks:
2017-12-21 17:31:52  Octoprint starts its internal GCODE startup to load filament
2017-12-21 17:34:43  Bed leveling and print start....  
2017-12-21 17:45:25  Send: M117 "OctoPi finish done"     

2017-12-21 17:59:11,100 - Recv: ok T:23.2 /0.0 B:42.9 /0.0 @:0 B@:0

Printer is cold. I click now the flash icon.
2017-12-21 18:00:00,098 - Send: M81
2017-12-21 18:00:01,144 - Recv: ok
2017-12-21 18:00:01,217 - Send: M105
2017-12-21 18:00:01,245 - Recv: ok T:23.2 /0.0 B:42.5 /0.0 @:0 B@:0
...

kantlivelong, 2017-12-21:
I do see in the logs that the job has stopped and indeed the idler timer isn't kicking in. So far I'm unable to replicate it here.

Can you try disabling all other 3rd party plugins and run again?

azalanono, 2017-12-21:
I have disabled several 3rd party plugins.
The only enabled plugins are now:

EEPROM Marlin Editor Plugin (1.0.4)
FileManager (0.1.2)
Fullscreen Plugin (0.0.4)
Navbar Temperature Plugin (0.9)
PSU Control (0.1.6)
This plugins are installed since I use octoprint. They cannot be the cause.

kantlivelong, 2017-12-21:
Please try this debug build.

1. Uninstall PSU
2. Restart OctoPrint
3. Install PSU Contro using this URL https://github.com/kantlivelong/OctoPrint-PSUControl/archive/issue_82_debug.zip
4. Retest with DEBUG logs and serial logging.
5. Send logs

azalanono, 2017-12-21:
[octoprint.log.zip](https://github.com/kantlivelong/OctoPrint-PSUControl/files/1580507/octoprint.log.zip)

kantlivelong, 2017-12-21:
Thanks. It looks as if the idle limit is reached before your hotend is ready and then afterwards never restarted. Can you confirm by trying a longer timeout and retry? I'll adjust my tests here to replicate.

azalanono, 2017-12-21:
[octoprint.log.zip](https://github.com/kantlivelong/OctoPrint-PSUControl/files/1580592/octoprint.log.zip)

kantlivelong, 2017-12-21:
Confirmed and also replicated here.

Thanks for the help.

azalanono, 2017-12-23:
Hi, Bruce.
I notice another situation which may be problematic for PSU control ...  Imagine that one clicks on "pause" to do some manipulation on the printer. I.e. change the filament.  This may also lead into a timeout on the idle timer when the printjob is continuing later.  
Also I currently see a massive mis-behavior between idle timer and temperature.

**Currently the PSU only turns off when the timer expired.**  If I set it to long, the extruder has reached 30°C and I still have all fans running until the timeout happens. When I set it to short time, PSU Control does not turn off the PSU.
The idle timer and monitoring the temperature **must** work independently from each other to establish proper operation

**I did some excessive analysis and want to share my thoughts with you. You will for sure notice the flexibility of this approach:**

1. **Use two idle timers**: "Heaters-Idle" is for automatically turning heaters-off, while "Command-Idle" is for detecting idle times between commands.  _Simply said, "Heaters-Idle" is a timeout for sending a "Turn heaters off" command when nothing happens for longer time, while "Command-Idle" is for inhibiting "Watch the temperature for switching the PSU" for a shorter time (i.e. 60 seconds)._ 

1. _New:_ **Watch data flow** and re-trigger **_both_** timers with whitelisted commands. _This is what you currently already do, only extended to two timers._ 

1. _New:_ **Watch the temperature** and the direction of its changing. As long as temperature is increasing **_and_** below 50°C, **set a "flag" to mark this condition**. The "flag" is cleared when temperature is above 50°C **_or_** the temperature is decreasing. _So every startup situation from "cold" can last as long as it needs, it is detected and marked by this "flag". Even when one clicks "Abort" or manually turns heaters off, this is correctly detected as decreasing temperature and the flag is cleared then._

1. _New:_ As long as the "flag" is set, re-trigger the "Heaters-Idle" timer repeatedly, do nothing else. _This is the safe way to prevent the timer from expiring. _

1. _New:_ **Watch Heaters-Idle timer** for switching heaters off.  When this timer expires, send commands to turn off the heaters. Nothing else has to happen. _So this part here turns off the heaters only when the printer appears to be idle but being heated. This in turn will finally lead into triggering the next point after some time. When heaters are already turned off from end of job commands, it does not have any effect to turn them off twice._ 

1. _New:_ **Watch the temperature for switching the PSU**.    As long as the temperature is above 50°C **_or_** the "flag" is set, do not turn the PSU off, otherwise turn the PSU off when "Command-Idle" has expired. _So when the temperature is below 50°C and no heatup phase, the PSU is turned off a short time after entering the last command._ 

I think this way PSU Control should correctly work in any situation. I might have overseen some detail, but from principle it should work.
- When sending a command without heatup, the PSU will stay ON for <Command-Idle> seconds, then turn off again.
- When the printer is heated up but idle (heaters haven't been turned off at printjob end), "Heaters-Idle" will trigger and turn heaters off by PSU-Control after some minutes.
- When printer cooling down, PSU will not be turned off until the printer is below 50°C.

kantlivelong, 2017-12-23:
Could you open a new ticket for this feature request (copy & paste) and I will review there. This is strictly a bug ticket that affects users with timeouts set to less than their heat-up time whilst using Set and Wait heating commands (M109 / M190).

azalanono, 2017-12-26:
I understand. Did so, the ticket is #85.

kantlivelong, 2017-12-26:
I've committed a fix if you would like to try it. Same URL (https://github.com/kantlivelong/OctoPrint-PSUControl/archive/issue_82_debug.zip)

long running commands will still cause the idle timer limit to be reached but the idle sequence will be skipped. On the next command the timer will be restarted.

Example:
G28 - Timer started/reset
M109 - Timer limit reached in cases where heatup > idle timer. In such cases the idle process skips and the timer is essentially left off until the next command. (This is the same behavior as 0.1.5)
G1 E1 - Timer started/reset

kantlivelong, 2017-12-29:
@azalanono have you had any chance to test this?

kantlivelong, 2018-01-01:
Considering this fixed as of 5808be2cacc1081ef34bc0b2bd80bf428a192b68. Closing

closed 2018-01-01 by kantlivelong