kantlivelong/OctoPrint-PSUControl #83: plugin conflict with TPLInk-Smartplug

jneilliii, 2017-12-24:
I've had a [user report](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues/36) that my code is interfering with gcode processing in your code, and I don't necessarily know why this would be the case. We're both using the gcode queuing hook to catch a specific command.  In my case a hard coded M80/M81 and in your case a configured gcode command.  Based on the documentation the way I am returning out of the hook function should be returning the gcode commands unaltered.  I was just curious if you have had any issues with this before or have any suggestions?

kantlivelong, 2017-12-26:
I'm not seeing anything obvious in your queuing hook that would cause an issue. I think both serial and debug(for both plugins) logs from the user would be required to see what is going on.

kantlivelong, 2018-01-02:
Closing due to inactivity.

closed 2018-01-02 by kantlivelong