kantlivelong/OctoPrint-PSUControl #84: PSU sensing blocks OctoPrint

btk5h, 2017-12-25:
I've been using PSUControl with [ouimeaux](https://github.com/iancmcc/ouimeaux/) to toggle a smart plug connected to my printer. PSUControl works perfectly fine with internal sensing, but system command sensing is completely unusable.

After setting the sensing command to `~/oprint/local/bin/wemo switch "Printer" status | grep -q 1`, OctoPrint blocks while waiting for the command to complete, effectively pausing the printer after every few instructions.

kantlivelong, 2017-12-26:
You may need to escape characters but I would suggest just putting it in a script file and calling that.

btk5h, 2017-12-26:
Sorry if my original issue was unclear. The system command works as-is; the issue is that PSUControl seems to be blocking OctoPrint while waiting for the system command to complete. The command takes a couple of seconds to query the status of the switch and during that time, OctoPrint can't send G-code to the printer. This causes the printer to stop frequently mid-print, creating some nasty print artifacts.

Not a Python guy, but I'm guessing that this line is responsible for the issue: https://github.com/kantlivelong/OctoPrint-PSUControl/blob/bc191bf8b32281dd763143dbb16ac40d26f71caf/octoprint_psucontrol/__init__.py#L344

kantlivelong, 2017-12-26:
That is intentional. Performing the command asynchronously would causes issues if sending GCODE that depend on power being on.

IE:
> M80
> G28

What are you switching on?

kantlivelong, 2017-12-26:
Scratch that. I read that as switching and not sensing. You are saying that using a script such as:

>#!/bin/bash
sleep 50

would hang OctoPrint?

btk5h, 2017-12-26:
Yes. Ideally, `check_psu_state` should poll the process using [`Popen.poll`](https://docs.python.org/2/library/subprocess.html#subprocess.Popen.poll) before calling `communicate`.

Maybe include a configurable timeout option to the plugin as well?

kantlivelong, 2017-12-26:
I've committed a fix to a temporary branch. Ended up using poll() and removing anything to do with stdout/stderr as it wasn't doing anything with it anyway.

You can install the test branch with the URL (Remove the existing version first):
https://github.com/kantlivelong/OctoPrint-PSUControl/archive/issue_84_fix.zip

btk5h, 2017-12-26:
https://github.com/kantlivelong/OctoPrint-PSUControl/blob/1f2107127bbe645473bdc3fdb2a3fd8c77865522/octoprint_psucontrol/__init__.py#L343-L345

This fix shouldn't work. `Popen.poll` doesn't block the thread waiting for the process to finish. Instead, it returns `None` if the process hasn't finished and `Popen.returncode` if the process has finished.

kantlivelong, 2017-12-26:
Hmm yeah I mean't to use wait() but it worked in testing. :man_shrugging:

Changed in 1df8998a74c86bc42dd4e004a5fbdf7dbd026896

kantlivelong, 2017-12-26:
Err nope. My head is broken today. Give me a bit.

kantlivelong, 2017-12-26:
Interesting. Even using poll() is blocking even though check_psu_state() is in its own thread.

See d3daa181072a83a66c6b791bd68422b3777d9528

kantlivelong, 2017-12-26:
Okay I am fairly confident that this latest change should resolve the issue as well as unify how all  system commands are executed.

Give it a shot and let me know how you make out.
https://github.com/kantlivelong/OctoPrint-PSUControl/archive/issue_84_fix.zip

PS:
I may add a configurable timeout as well as polling interval at a later date.

kantlivelong, 2017-12-29:
Confirmed working. Merged into devel with commit 36b1282d0aaf33912ad78b61fd438a09f2e6c7ee

closed 2017-12-29 by kantlivelong