kantlivelong/OctoPrint-PSUControl #87: PSU switches not detected

MasterPIC, 2018-01-04:
Hi,
I'm using (on Pi Zero W) in mode and out mode to control a PSU (by gpio pin #23 ), so I think I should use GPIO float as sensing method.
Anyway the plugin doesn't sense the PSU state as expected.
In my case PSU is ON in out mode, OFF in input mode. Playing with revert checkbox doesn't solve the issue.

I tried also to use system command (gpio -g read 23) as sensing method, still unsuccessfully.
So I'm tempted to think something related to polling is buggy...
I'm using PSU Control 0.1.7. I'd like to avoid internal sensing since an external PSU control would make sensing not working...

kantlivelong, 2018-01-04:
Could you provide:
1. A wiring schematic (crude is fine)?
2. Your PSUControl configuration. (screenshot is fine)

MasterPIC, 2018-01-04:
Hi @kantlivelong 
there is an optocoupler, a MOSFET, some resistors and a relay. A schematic is not worth it...
Simply in mode and out mode (on pin n.23) are used to switch ON or OFF the PSU.
If PSUControl thinks PSU is ON and PSU is really ON I can switch PSU OFF by the toggle button.
If PSUControl thinks PSU is OFF and PSU is really OFF I can switch PSU ON by the toggle button.
What isn't working is the PSU state tracking (and I tried several configurations)...
gpio -g read 23 returns 1 or 0 as expected by PSUControl, according to PSU state. So system command sensing should work, but it doesn't.
Maybe the plugin doesn't execute the sensing check (to track PSU state) at all... What is the polling time?

Thanks again!
Armando

kantlivelong, 2018-01-04:
System command works based on the exit code. 0=on,1=off.

Will need the details requested to support any further.

MasterPIC, 2018-01-05:
Hi @kantlivelong 
I thought **gpio -g read 23** was a good command to sense PSU state (by opting for system command as sensing method).
I solved the issue by making a script containing the following code:
```
test="$(gpio -g read 23)"
if [ "$test" -gt 0 ]
 then
 exit '1'
fi
exit '0'
```
Sorry for the annoyance I caused.

Thanks again
Armando

kantlivelong, 2018-01-05:
Just out of curiosity. Why aren't you using native GPIO support?

MasterPIC, 2018-01-05:
Because of the ON state at power on, instead of modifying the PCB I opted for switching between in mode and out mode. Probably this is the reason why reading the logic level on the output wouldn't work in my case.

kantlivelong, 2018-01-05:
ON state of the sense pin? The default state of the GPIO pins on boot should be set as inputs and not outputs. See https://raspberrypi.stackexchange.com/a/514

MasterPIC, 2018-01-05:
This is the schematic of the board used to control the PSU:
[image]
On power on by keeping the pin in input mode the PSU stays OFF. I can switch PSU ON by enabling out mode (I assume 0 is the default output level).
Do you see something wrong?

Thanks again!

kantlivelong, 2018-01-05:
Where are you connecting your GPIO pins?

In1 = Switching GPIO Pin?
??? = Sense?

MasterPIC, 2018-01-05:
In1 is connected to GPIO n.23; no other connection are in use to control or sense the PSU.

kantlivelong, 2018-01-05:
Based on the schematic you need the GPIO pin to act as open-drain which essentially will tie _In1_ to ground when the GPIO pin is high. Raspberry Pi's do not have open-drain circuitry built in but its easy to do. See [this example](https://upload.wikimedia.org/wikipedia/commons/thumb/c/c0/OpencollectorV3.png/1200px-OpencollectorV3.png)

As a simple test try connecting In1 to a common ground. When connected the PSU should turn on.

kantlivelong, 2018-01-05:
... You should also stick to Internal Sensing until that is worked out.

MasterPIC, 2018-01-05:
I'm not sure but I think CMOSlogic is used by Raspberry on the outputs. By limiting the current flowing into the Raspberry I see no risks by using the low output to switch ON the PSU... Am I wrong?

kantlivelong, 2018-01-05:
LOW is not tied to ground but instead floating. Essentially it's like not being connected.

MasterPIC, 2018-01-05:
But it works. So I think it's tied to ground...

kantlivelong, 2018-01-05:
HIGH = 3.3v
LOW = 0V

0V != GND

It may "work" but it's not right and you may kill the IC.

http://rpi.science.uoit.ca/lab/gpio/
https://electronics.stackexchange.com/a/76310

kantlivelong, 2018-01-05:
It looks like you are using a relay similar to [this](https://www.sainsmart.com/products/4-channel-5v-relay-module)

Here's a thread about it: https://www.raspberrypi.org/forums/viewtopic.php?t=197339

MasterPIC, 2018-01-05:
I know sink current flowing into GPIO pins should be less than 16mA.
On the relay board I'm using (like the one you linked to) because of the 3.3V voltage, in place of the 5V for which the board was designed, I opted for removing the 2nd led on the input port of the optocoupler to properly control it (instead of changing the resistor). The resulting current is well under the limit of 16mA: I remember it's about 2mA (I took into account the optocoupler specs together with the Raspberry ones). So I see no reasons to worry about.
What I'm not sure about is related to native GPIO support. What do you mean with it?

As for the internal sensing, if I manually send a system command to change the PSU state then PSUControl doesn't sense it...

kantlivelong, 2018-01-05:
PSUControl has built-in GPIO support for the Raspberry Pi so you shouldn't need to use commands and can instead specify what pin number to use.

MasterPIC, 2018-01-05:
OK, I see that opting for GPIO Pin as switching method PSUControl properly switches ON/OFF the PSU.
If I specify 23 as pin number I've to set BCM numbering.
But I see that PSUControl takes full control of the PSU if I don't use system commands to control and sense PSU.
In a few words  I'd like to be able to keep control of PSU not only by PSUControl, also by terminal and system commands
Could you tell me how to set PSUControl??

Thanks again!

kantlivelong, 2018-01-05:
Sorry but I am not sure what you are saying. Can you rephrase?

MasterPIC, 2018-01-05:
Selecting GPIO Pin method in PSUControl (to control &/OR sense PSU state) I'm no more able to control externally the PSU: PSUControl takes full control of the PSU.

kantlivelong, 2018-01-05:
externally how? If you have the switch pin wired to the pi then yes the pi would control it.

MasterPIC, 2018-01-05:
"gpio -g mode 23 out" (or in) by terminal to control the PSU (without using PSUControl)...
Or a click from the browser on the system command used to control the PSU...
Or by Telegram (to execute the system command able to control the PSU)...

In a few words as soon as I select GPIO Pin method only PSUControl can control my PSU.

kantlivelong, 2018-01-05:
So you turn it on by changing the pin to out or in?

MasterPIC, 2018-01-05:
yes

kantlivelong, 2018-01-05:
That is because you have it wired incorrectly.

MasterPIC, 2018-01-05:
The input port of an optocoupler is connected to Raspberry GPIO and its powered by Raspberry itself at 3.3V. There is a resistor (1kOhm if I remember correctly). The pulldown device inside Raspberry can manage the consequent current.

Switching to OUT mode makes the PIN not connected to ground and the photocoupler works as expected (no current can flow to ground on the input port of the optocoupler).

Maybe PSUControl has not been designed to work with my control board...

MasterPIC, 2018-01-05:
I just edited my previous answer about the obvious pin modes required to properly control the optocoupler.

sbts, 2018-01-05:
@MasterPIC Rather than editing, can you please "clearly repost" what your intent is.
It's extremely hard to follow a conversation like this so we can assist you if we have to hunt back in history for information.

That said, The code as it exists is known to work in may situations.
To help you it would be useful if you can please redraw an supply an accurate Schematic and Wiring information (Schematic shows the board and it's inputs and output, wiring shows how it connects to the Pi and the PSU)

I suspect your issues are partly related to how you have things wired, and partly a Language issue.

I'm happy to assist, but will need the requested information, including screenshots of how you have the plugin configured

MasterPIC, 2018-01-05:
I just opened a new issue #88

closed 2018-01-05 by MasterPIC